rerdavies/pipedal #307: [BUG] Error compiling on Fedora

djshaji, 2025-05-18:
**Describe the bug**
If I try to compile the project on Fedora, I get the following error:
`[ 38%] Building CXX object src/CMakeFiles/namSources.dir/namFixes/get_dsp.cpp.o
cd /home/djshaji/projects/ToobAmp/build/src && /usr/bin/c++ -DNAM_SAMPLE_FLOAT -I/home/djshaji/projects/ToobAmp/src/../modules/NeuralAmpModelerCore -I/home/djshaji/projects/ToobAmp/src/../modules/NeuralAmpModelerCore/Dependencies/eigen -I/home/djshaji/projects/ToobAmp/src/../modules/NeuralAmpModelerCore/Dependencies/nlohmann -Wno-psabi -Werror -Wall -pedantic -Wno-reorder -Wno-restrict  -Ofast -DNDEBUG  -flto=auto -fno-fat-lto-objects -fvisibility=hidden -std=gnu++20 -fPIC -Wno-pedantic -Wno-deprecated-declarations -Wno-unused-but-set-variable -Wno-deprecated-enum-enum-conversion -Wno-sign-compare -Wno-effc++ -Wno-error -MD -MT src/CMakeFiles/namSources.dir/namFixes/get_dsp.cpp.o -MF CMakeFiles/namSources.dir/namFixes/get_dsp.cpp.o.d -o CMakeFiles/namSources.dir/namFixes/get_dsp.cpp.o -c /home/djshaji/projects/ToobAmp/src/namFixes/get_dsp.cpp
In file included from /home/djshaji/projects/ToobAmp/src/../modules/NeuralAmpModelerCore/Dependencies/nlohmann/json.hpp:39,
                 from /home/djshaji/projects/ToobAmp/src/../modules/NeuralAmpModelerCore/NAM/dsp.h:13,
                 from /home/djshaji/projects/ToobAmp/src/namFixes/get_dsp.cpp:17:
/usr/include/c++/15/ciso646:46:4: warning: #warning "<ciso646> is deprecated in C++17, use <version> to detect implementation-specific macros" [-Wcpp]
   46 | #  warning "<ciso646> is deprecated in C++17, use <version> to detect implementation-specific macros"
      |    ^~~~~~~
In file included from /home/djshaji/projects/ToobAmp/src/namFixes/get_dsp.cpp:23:
/home/djshaji/projects/ToobAmp/src/namFixes/wavenet_t.h: In member function ‘long int nam::wavenet::_Layer_T<INPUT_SIZE, HEAD_SIZE, CHANNELS, KERNEL_SIZE>::get_channels() const’:
/home/djshaji/projects/ToobAmp/src/namFixes/wavenet_t.h:208:91: error: ‘const class nam::wavenet::_Layer_T<INPUT_SIZE, HEAD_SIZE, CHANNELS, KERNEL_SIZE>’ has no member named ‘_conv’ [-Wtemplate-body]
  208 |   long get_channels() const { return _gated ? this->_conv_gated.get_in_channels() : this->_conv.get_in_channels(); };
      |                                                                                           ^~~~~
gmake[2]: *** [src/CMakeFiles/namSources.dir/build.make:166: src/CMakeFiles/namSources.dir/namFixes/get_dsp.cpp.o] Error 1
gmake[2]: Leaving directory '/home/djshaji/projects/ToobAmp/build'
gmake[1]: *** [CMakeFiles/Makefile2:1939: src/CMakeFiles/namSources.dir/all] Error 2
gmake[1]: Leaving directory '/home/djshaji/projects/ToobAmp/build'
gmake: *** [Makefile:169: all] Error 2
`

** Environment (please complete the following information):**
OS - Fedora release 43 (Rawhide)
uname -a:  Linux wanderer 6.15.0-0.rc0.20250404gite48e99b6edf4.11.fc43.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Apr  4 20:17:28 UTC 2025 x86_64 GNU/Linux

I think it might be related to some version upgrade of the NAM plugin, as NAM is pulled directly off of it's git repo.

Thanks and regards

rerdavies, 2025-05-18:
The submodule dependency on NAM should be pinned to a particular tag, so I
don't think that's it. The error is actually a warning related to
depreciation of an ancient C++ in C++/20; but none of my test compilers
give me the warning.

It's more probably related to compiler version. Which compiler/which
version are you using?

For your reference, the project is currently built on GCC 12.x,or 13.x and
is actually built on a Clang compiler as part of GitHub CI (not sure which
version of the to of my head).

What compiler/ version are you using?

djshaji, 2025-05-21:
I am using:
- gcc version 15.0.1 20250329 (Red Hat 15.0.1-0) (GCC) 
- clang version 20.1.2 (Fedora 20.1.2-3.fc43)

I've only tried to use gcc as of now. I'll try with clang too. I use C/C++, but haven't used templates much, so couldn't debug it myself.

Thanks and regards,