rerdavies/pipedal #312: Disable automatic WIFI Scan

jpbarraca, 2025-05-31:
On a RPI 5, using the Internal Wifi and a I2S DAC, there is noise when a Wifi Scan takes place.

Is it possible to disable the periodic Wifi Scan while associated to an Access Point?

The drawback will be loss of support for Roaming, but that may not be critical for all scenarios

rerdavies, 2025-06-04:
> Hi,
>
> 1: no noise with wifi traffic. Even with high load (ping -f, iperf,
> browser)
> 2: no noise sending or receiving broadcast messages (using udp traffic to
> network broadcast address at line rate)
> 3: the wifi card is connected to an AP, 2 meter away, channel 100, width
> 80mhz, auto tx power, power save disabled. I'm not using the hotspot mode.
> fixed tx power, high tx power, low txt power is the same.
>
> Raspberry Pi 5 Model B Rev 1.0
> Linux pipedal 6.12.25+rpt-rpi-2712
>
> It can be easily triggered with iw wlan0 scan
>
> Attached you can find the sound as a result of forcing a scan with iw
> noise.zip <https://github.com/user-attachments/files/20581779/noise.zip>

Ok. I can fix that. May take a few days (3 or 4) to issue because I have
some large changes in flight right now. Expect it in the next posted
release.

jpbarraca, 2025-06-04:
Thanks for the update. @SkyhawkBS do you know if the issues persist with an external USB dongle?

jpbarraca, 2025-06-04:
Agree that the issue may be with DC offsets.  I can test with a USB wifi card, and a USB audio interface. 
The purpose is to build a self contained pedal, which already exists and works well, except for this issue.

I recognize that the issue is not due to pipedal, and this will not be a real fix, but it will provide more control and allow users to avoid the issue. Isolating the I2S interface will be the next level to follow.... actually, all the rest is already DC isolated because of similar issues.

rerdavies, 2025-06-04:
Excellent. I will make the change anyway. PiPedal should not be turning on the Wifi if it doesn't have to. And PiPedal should not be doing scans if it doesn't have to. (But that won't solve your problem. lol)

Sounds like you know what you're doing.  (audio traces routed between ground planes, and extra board layers to allow for that used to be the rule back in the day when I last worked with a pro-audio component, which was a very long time ago).

closed 2025-07-18 by jpbarraca