rerdavies/pipedal #317: Allow calibration of min/max expression pedal values

briannadon, 2025-06-25:
Hi there, our back-and-forth regarding getting pipedal to work with bluetooth MIDI devices was very enlightening, and led me to this suggestion:

I would love a feature to calibrate the min/max value of an expression pedal. Most (or I think basically all) multifx pedals that accept expression pedals allow for calibration of the expression pedal, because most expression pedals don't *quite* reach 0 -> 127 due to hardware limitations. For example, mine goes from about 13 -> 127 (per amidi --dump). 

I would love the ability to have pipedal listen for the min and max value of a MIDI channel, and then map the minimum to 0.0 and the maximum to 1.0 (in e.g. linear mode). As of now, I can't use my expression pedal as a whammy because of this.

rerdavies, 2025-07-01:
Implemented  in version v1.4.77 (just published).

briannadon, 2025-07-01:
Awesome! Tested and works nicely. Thanks!

closed 2025-07-01 by briannadon