rerdavies/pipedal #333: [BUG] debug mode pipedald.service terminated by SIGABRT

jardous, 2025-07-14:
pipedald.service gets terminated in debug mode:

```
Jul 14 22:23:57 raspberrypi pipedald[14024]: HotspotManager: state=Monitoring
Jul 14 22:23:57 raspberrypi pipedald[14024]: HotspotManager: Monitoring network status.
Jul 14 22:23:57 raspberrypi pipedald[14024]: HotspotMonitor: ethernetConnected=0
Jul 14 22:23:57 raspberrypi pipedald[14024]: HotspotMonitor: OnWlanStateChanged
Jul 14 22:23:57 raspberrypi pipedald[14024]: Scanning
Jul 14 22:23:57 raspberrypi pipedald[14024]: terminate called after throwing an instance of 'std::bad_alloc'
Jul 14 22:23:57 raspberrypi pipedald[14024]:   what():  std::bad_alloc
Jul 14 22:23:57 raspberrypi systemd[1]: pipedald.service: Main process exited, code=killed, status=6/ABRT
Jul 14 22:23:57 raspberrypi systemd[1]: pipedald.service: Failed with result 'signal'.
Jul 14 22:24:03 raspberrypi systemd[1]: pipedald.service: Scheduled restart job, restart counter is at 18.
Jul 14 22:24:03 raspberrypi systemd[1]: Stopped pipedald.service - PiPedal Web Service.
Jul 14 22:24:03 raspberrypi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Jul 14 22:24:03 raspberrypi pipedald[14056]: Document root: /etc/pipedal/config Threads: 5
```

rerdavies, 2025-07-14:
Huh. Which compiler? Which OS? 

Give me 20 minutes. If you're rolling your own builds, I will check in the fix for the MIDI select on a debug branch.

rerdavies, 2025-07-14:
See note on the MIDI controller issue (fix is checked in on the dev-modui branch).

IF you want to pursue this... make sure the Website sources have been updated as well as the binary. If they don't match, bad things may happen. The Pipedal web sources are, for reasons of expediency and debug turn time, NOT built in a debug build.  

To fix this, you can do either of the following: 

1) Run a full RelDebugInfo (or release) buid, which will also build the PiPedal web content. Run ./install.sh in the project root directory. 

2) 
```
   cd vite
   ./build.sh  #builds the Pipedal web content.
   cd ..
   ./install.sh    # copies into the service directories.
```

Feeling rather badly about all the trouble I'm putting you through. :-/

rerdavies, 2025-07-14:
btw, your favicon pull didn't stick. :-( The docs for this project use the default github jeckyll docs build procedure, which .... is poisonously horrible and useless. Don't EVER use jeckyll. Since then, I've learned how to build and serve vite/react websites from a github page. All my other sites use this scheme. But converting the Pipedal docs is a large project, unfortunately. And I'm stuck with Jeckyll for the forseeable future.

Bottom line: I CANNOT for the life of me figure out how to get the Github build pages to insert the <link /> for the favicon. (Pngs work better, apparently, Firefox doesn't like .ico files).

rerdavies, 2025-07-20:
Not able to replicate in the current dev build. Possibly fixed.

rerdavies, 2025-07-22:
I'm going to provisionally close this, as I am not able to reproduce at present. If you see it again, let me know. And we'll do the `valgrind` thing.