rerdavies/pipedal #337: Audio Adapter Crashes

rerdavies, 2025-07-16:
Reports of two audio adapters that crash when used with PiPedal.

_Originally posted by @jardous in https://github.com/rerdavies/pipedal/discussions/329#discussioncomment-13776353_

@jardous:

My bad. Run the following command, and send me the results. You can trim the file if it's been a while since you last rebooted.  Or you can replicate the issue after a clean reboot, and send the complete results (preferable). 

```
   journalctl -b0 | grep pipedald > tmp.txt
```

I am -- to be honest -- surprised to see any sort of crashes with USB audio adapters, after having tested with a fairly broad range of adapters. And this one looks like it's worth saving. Clues as to where the crash occurs might help.

If the logs show that PiPedal is exiting without any indication of a clean shutdown sequence, or a SIGSEGV fault, you might want to take a peek in the journalctl logs around the point at which the process terminates to see if there are indications of a more severe crash.

As previously mentioned, there are outstanding issues with ALSA which sometimes needs a system reboot after adding or removing USB adapters while running. (Also sometimes after severe underruns, although current versions of ALSA seem to have fixed that). So a clean reboot might be the best way to go. ALSA doesn't seem to entirely like it when a client process terminates abnormally, and occasionally goes into a state where nothing works until the system is rebooted.

rerdavies, 2025-07-20:
I would like to impose on your developer fu, if you don't mind. The following procedure is a little more extreme than I would use with non-developers. But it should precisely determine what's going on with your two malfunctioning USB adapters. If you need more details on any of the following steps, don't hesitate to ask.

The Positive Grid RIFF device seems like a respectable device, that's eminently worth saving,  so let's make that a priority. If you want to take the extra time to do the same for the ZOOM pedal, I'll take those results as well, but I'm suspecting the underlying issue is the same with both devices. (Maybe mono/mono port configuration (I don't think so), or maybe USB 1.0 devices instead of USB 2.0 devices (although I don't think so).  Or maybe a plugin that's loading at startup (I rather do think so). We shall know when we are done.

What I would like to do is run `valgrind` to figure out what's going on with your audio devices. To that end, I have prepared an experimental release with full debug symbols, and debug-optimized code. That will allow `valgrind` to produce useful stack traces that will allow me to identify the root cause pretty precisely. 

## Preparatory Step

Install `valgrind`:

    sudo apt install valgrind

Add your userid to the `pipedal_d` group (which allows your user-id to read and write to pipedal's configuration and storage directories. 

    sudo usermod -a -G pipedal_d [YOUR USER ID]

Reboot the system to get the change to take effect.

## Installing the Experimental buid.

The experimental build has been published here:

    https://github.com/rerdavies/pipedal/releases/tag/v1.4.78

Download the .deb package and install it by running: 

    sudo apt-get install ./pipedal_1.4.78_arm64.deb

You MUST use `apt-get`; and the leading `.` in the filename is not optional. You can ignore the warning. 

## Running the test.

Stop the pipedald service:

     sudo systemctl stop pipedald

Run the `pipedald` executable with `valgrind`

    valgrind  --log-file=~/valgrind.output /usr/bin/pipedald /etc/pipedal/config /etc/pipedal/react -port 0.0.0.0:8080 -log-level debug

`valgrind` diagnostics will be written to the file `~/valgrind.output`.  PiPedal will display log messages on your terminal. If it terminates abnormally, then we should have fairly precise information on where the crash occurred. If it launches successfully, then the pipedal service will be running (although abysmally slowly, and underrunning furiously).  However, it seems to host the web pages without problems. Fire up a browser, and configure PiPedal to use one of your recalcitrant USB audio devices.  The pipedal web UI will be running on port 8080. 

Once done, if `pipedald` hasn't crashed, you can terminate the running instance by typing Ctrl+C in the console. PiPedal will do an orderly shutdown. If there are any problems during shutdown, I'd be interested in those as well.

Pipedal's log messages are NOT captured in the valgrind file. If you could also cut and paste log messages  immediately preceding an abnormal termination, that would be helpful.

If everything goes well, then the problem is most likely fixed. If the server has crashed, we will have extremely useful info in the `~/valgrind.output` file.  Succesful or not, email me the contents of the ~/valgrind.output` file (or post it here).

## Restoring Pipedal to Usable State Afterward.

Download the most recent build of pipedal from 

    https://github.com/rerdavies/pipedal/releases/tag/v1.4.77

Run the following commands:

     sudo apt remove pipedal  #uninstalls the experimental build.
     sudo apt-get install ~/Downloads/pipedal_1.4.77_arm64.deb

PiPedal does not remove or modify your configuration data when uninstalling; and -- although there are extra configuration properties in preset files, they SHOULD downgrade gracefully. There are no other changes in configuration files. If there are any problems, let me know, and I will cut you an emergency non-debug build immediately. To be safe, it would be best if you do not SAVE an presets while using the experimental build.

Regards,

R.

jardous, 2025-07-20:
Thanks for the detailed description. However I have already sent the interface back. As I have already experience more, that have similar issue, I am pretty sure, that this will come in handy.

rerdavies, 2025-07-22:
If anyone can give me a repro on this issue, please let me know. The good news: at least I have a technique for running the issue down. Now all I need is somebody who has an adapter that reproduces this issue.

rerdavies, 2025-07-27:
> are you sure you get stack traces with valgrind? Shouldn't GDB be used
> instead?

You do get very nice stack traces with valgrind, and additionally valuable
tests for memory corruption, as long as everything runs fast enough. As it
turns out, it more-or-less works out. gdb would be a good approach too if
valgrind is too slow.

closed 2025-08-12 by rerdavies