rerdavies/pipedal #340: UI remembers open plugin for each preset

DuraMorte, 2025-07-18:
Hello again!

I would like for PiPedal to remember which plugin window was open in the GUI when I change the preset with my MIDI foot controller.
I have a "tuner-only" preset, that mutes all outputs and only lets in the guitar to feed the tuner.
But sometimes, when I mess around with plugins and values in other presets, then switch back to my tuner-only preset, I don't get the tuner GUI back, but the Split plugin.

I realize that I could just remove the split and leave just the tuner plugin only in that preset, but the same thing happens in every other preset; I modify a parameter in a plugin on one preset, save the change, then swap back and forth between that preset and some others. When I swap back to the preset that I'm modifying, the plugin window that I was modifying previously probably won't be the one I see. It might be, but it is unreliable.

If this isn't possible, it's easy to work around. Just a small tweak that would make my life easier. :)

rerdavies, 2025-07-19:
Not a bug at all. PiPedal never did this. But now it will, in the next release. :-) 

TLDR: Changing the plugin selection does NOT mark the current preset as changed -- it seems to trivial and common to justify marking the preset as changed . However, the Save icon is now always visible, so you can change the preset, click save, and the selection will be restored when you reload.  Of course the selection gets saved if the preset has changed and you save, too.

DuraMorte, 2025-07-19:
If the selected plugin saves with the preset, that is as elegant a solution as I could ask for.

Thank you!

rerdavies, 2025-07-22:
Fixed in v1.4.79.

closed 2025-07-30 by rerdavies