rerdavies/pipedal #343: [BUG] Error loading Plugins with MIDI Event Ports

jpbarraca, 2025-07-20:
**Describe the bug**

With Calf Plugins (https://github.com/calf-studio-gear/calf), pipedald reports an error when adding a new element to  the chain. The error happens when clicking the + button and a Calf plugin already exists in the chain.
Because this relates to external plugins, I'm not sure if it is a bug or enhancement. It may be useful to be registered.

The error is the direct display of the message send through the websocket:
```
[
    {
        "message": "error"
    },
    "Expecting '{', near: ',\"midiDevices\":[],\"channel\":-1,\"acceptPr'."
]
```

The error is created with plugins that have these ports declared. Removing the definitions enables the use of the plugin.

```
    [
        a lv2:InputPort ;
        a atom:AtomPort ;
        lv2:index 20 ;
        lv2:symbol "events_in" ;
        lv2:name "Events" ;
        lv2:portProperty lv2:connectionOptional ;
        atom:bufferType atom:Sequence ;
        atom:supports lv2midi:MidiEvent ;
        atom:supports atom:Property ;

    ] , [
        a lv2:OutputPort ;
        a atom:AtomPort ;
        lv2:index 21 ;
        lv2:symbol "events_out" ;
        lv2:name "Events" ;
        lv2:portProperty lv2:connectionOptional ;
        atom:bufferType atom:Sequence ;
        atom:supports lv2midi:MidiEvent ;
        atom:supports atom:Property ;

    ]
```

** Environment:**
 - OS: Linux pipedal 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
 - RPI 5 8G
 - PiPedal v1.4.77-Beta

[Gate.ttl.txt](https://github.com/user-attachments/files/21333606/Gate.ttl.txt)

rerdavies, 2025-07-20:
Fix checked in. Will be available in an imminent forthcoming release.

jpbarraca, 2025-07-22:
Validated that it is fixed in v1.4.79. 
Calf plugins loading without errors.