rerdavies/pipedal #346: Hailo AI HW accelerator

jardous, 2025-07-22:
Hello,

I was thinking about if this would be worth a try. The Hailo module costs about 100 bucks.

Initial research shows, that if we were able to convert the NAM model to Hailo format, we could use the Hailo SDK compiler to run the NAM plugin accelerated.

What are you thoughts about this? Would that bring any real improvement worth the hassle?

My motivation would be to be able to use even smaller buffers without xruns.

rerdavies, 2025-07-22:
Are you getting buffer overruns on small buffer sizes? I don't think you should be. I can run 32x4, and stack up my NAM instances until the CPU use is nudging 85%, with rock solid zero-overruns. Which is kind of extraordinary. 85%! I'm used to windows where you have to run 20-30% CPU load to avoid overruns. (Fwiw,  I think I may have a bug which prevents TooB NAM from running properly on 16x buffers, that I really should fix. I've seen  a field report to that effect. TooB NAM does buffer assembly when running on frames that are less than 32x, so it's quite possibly true. i.e. TOOB Nam actually runs with slightly more latency when you are running 16x buffer confiigurations. :-/ .

Re the AI modules. I think it would be a pretty high risk endeavor.  Challenges off the top of my head:

- Whether you can get data back and forth between a coproessor and a CPU on a realtime thread. 
- Whether the TPU is really going to like the tiny models that NAM uses. 
- The lingua franca that all TPUs speak is python!! Which isn't going to run in realtime.As far as I know, the only APIs for TPUs that are broadly supported by a wide variety of TPUs in a wide variety of platforms is a Python API.

What TPUs really like is big huge matrix multiplies. Oddly, the neural net models that NAM uses are very small. I vaguely remember the vast majority of compute time is spent on tiny 24x32 (or something in that order) matrix multiples used in one of the foundation models (a bunch of small ones, and one big one) . That's actually the foundation premise of NAM (and it's predecessor ML). So it's not immediately obvious that a TPU is going significantly improve NAM's performance. It might. It might not. 

My estimate, based on experience with the code: a very significant 3 to 6 month development effort. With significant risk of failure. And not a whole lot of benefit, given that you SHOULD (I think) be able to run 5 NAM instances in a 32x3 or 32x4 buffer configuration on a Pi 5.

rerdavies, 2025-07-22:
26 TOPs for $100 is pretty extraordinary though. I might actually shell out for one of those.