rerdavies/pipedal #347: Thread-per-plugin-instance LV2 Worker threads

rerdavies, 2025-07-22:
Match the MOD threading model in order to allow the MOD NAM plugin to work.

_Originally posted by @rerdavies in https://github.com/rerdavies/pipedal/discussions/339#discussioncomment-13853849_

rerdavies, 2025-07-23:
@StuSysMoo

Discussion moved to an issue thread, and out of the main discussion page.

Bringing out the heavy guns. The following procedure is a bit awkward, but it's probably the best way to run the issue down.  I'm more than a bit concerned that we will end up determining that it's a bug in the PatchStorage pluing; but at least we will know.

What I would like to do is run `valgrind` to figure out what's going on with your audio devices. To that end, I have prepared an experimental release with full debug symbols, and debug-optimized code. That will allow `valgrind` to produce useful stack traces that will allow me to identify the root cause pretty precisely. 

## Preparatory Step

Install `valgrind`:

    sudo apt install valgrind

Add your userid to the `pipedal_d` group (which allows your user-id to read and write to pipedal's configuration and storage directories. 

    sudo usermod -a -G pipedal_d [YOUR USER ID]

Reboot the system to get the change to take effect.

## Installing the Experimental buid.

The experimental build has been published here:

    https://github.com/rerdavies/pipedal/releases/tag/v1.4.81

Download the .deb package and install it by running: 

    cd ~/Downloads
    sudo apt-get install ./pipedal_1.4.81_arm64.deb

You MUST use `apt-get`; and the leading `.` in the filename is not optional. You can ignore the warning. 

## Running the test.

Stop the pipedald service:

     sudo systemctl stop pipedald

Run the `pipedald` executable with `valgrind`

    valgrind  --log-file=~/valgrind.output /usr/bin/pipedald /etc/pipedal/config /etc/pipedal/react -port 0.0.0.0:8080 -log-level debug

`valgrind` diagnostics will be written to the file `~/valgrind.output`.  PiPedal will display log messages on your terminal. If it terminates abnormally, then we should have fairly precise information on where the crash occurred. If it launches successfully, then the pipedal service will be running (although abysmally slowly, and underrunning furiously).  However, it seems to host the web pages without problems. Fire up a browser, and configure PiPedal to use one of your recalcitrant USB audio devices.  The pipedal web UI will be running on port 8080. 

Once done, point a browser (or the Android PiPedal Remote app) at your Raspberry Pi. The UI will work, but it will be significantly slower than usual. You may have  to be patient, particularly when loading models. 

Do what you can to reproduce the issue. 

After Pipedal crashes, the `valgrind` program should terminate. If it doesn't you can type Control+C to make it terminate.

Send me the contents of the `~/valgrind.output` file.

## Restoring Pipedal to Usable State Afterward.

Download the most recent build of pipedal from 

    https://github.com/rerdavies/pipedal/releases/tag/v1.4.80

Run the following commands:

     sudo apt remove pipedal  #uninstalls the experimental build.
     sudo apt-get install ~/Downloads/pipedal_1.4.80_arm64.deb

PiPedal does not remove or modify your configuration data when uninstalling; so your data will and configuration will be preserved. 

Fingers crossed.

StuSysMoo, 2025-07-23:
getting some syntax errors in that command:  patch@Neutron-Pi:~/Downloads $ valgrind  --log-file=~/valgrind.output /usr/bin/pipedald /etc/pipedal/config /etc/pipedal/react -port 0.0.0.0:8080 -log-level debug
valgrind: /usr/bin/pipedald: is a directory

rerdavies, 2025-07-23:
Try this instead:
```
cd ~
valgrind  --log-file=valgrind.output /usr/sbin/pipedald /etc/pipedal/config /etc/pipedal/react -port 0.0.0.0:8080 -log-level debug
```

StuSysMoo, 2025-07-23:
ok i think i got it, not 100% qs it crashed a few times starting up the command but did then get it running and cause the crapout.. let me know if its no good i will try again.

[valgrind.output.txt](https://github.com/user-attachments/files/21377499/valgrind.output.txt)

StuSysMoo, 2025-07-23:
5
2025-07-23 02:15:18.256 INF: lilv: 2025-07-23 02:15:16.935 DBG: Plugin: PurestConsole2Buss
2025-07-23 02:15:18.256 INF: lilv: 2025-07-23 02:15:16.944 DBG: Plugin: PurestConsole2Channel
2025-07-23 02:15:18.256 INF: lilv: 2025-07-23 02:15:16.952 DBG: Plugin: PurestConsole3Buss
2025-07-23 02:15:18.257 INF: lilv: 2025-07-23 02:15:16.961 DBG: Plugin: PurestConsole3Channel
2025-07-23 02:15:18.257 INF: lilv: 2025-07-23 02:15:16.969 DBG: Plugin: PurestConsoleBuss
2025-07-23 02:15:18.257 INF: lilv: 2025-07-23 02:15:16.977 DBG: Plugin: PurestConsoleChannel
2025-07-23 02:15:18.257 INF: lilv: 2025-07-23 02:15:16.988 DBG: Plugin: PurestDrive
2025-07-23 02:15:18.257 INF: lilv: 2025-07-23 02:15:17.000 DBG: Plugin: PurestFade
2025-07-23 02:15:18.257 INF: lilv: 2025-07-23 02:15:17.012 DBG: Plugin: PurestGain
2025-07-23 02:15:18.258 INF: lilv: 2025-07-23 02:15:17.025 DBG: Plugin: RawGlitters
2025-07-23 02:15:18.258 INF: lilv: 2025-07-23 02:15:17.037 DBG: Plugin: RawTimbers
2025-07-23 02:15:18.258 INF: lilv: 2025-07-23 02:15:17.046 DBG: Plugin: Recurve
2025-07-23 02:15:18.258 INF: lilv: 2025-07-23 02:15:17.070 DBG: Plugin: ResEQ
2025-07-23 02:15:18.258 INF: lilv: 2025-07-23 02:15:17.082 DBG: Plugin: ResEQ2
2025-07-23 02:15:18.258 INF: lilv: 2025-07-23 02:15:17.093 DBG: Plugin: Reverb
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.102 DBG: Plugin: RightoMono
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.117 DBG: Plugin: SampleDelay
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.129 DBG: Plugin: ShortBuss
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.139 DBG: Plugin: SideDull
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.149 DBG: Plugin: Sidepass
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.159 DBG: Plugin: Sinew
2025-07-23 02:15:18.259 INF: lilv: 2025-07-23 02:15:17.169 DBG: Plugin: Slew
2025-07-23 02:15:18.260 INF: lilv: 2025-07-23 02:15:17.178 DBG: Plugin: SlewOnly
2025-07-23 02:15:18.260 INF: lilv: 2025-07-23 02:15:17.186 DBG: Plugin: Spiral
2025-07-23 02:15:18.260 INF: lilv: 2025-07-23 02:15:17.203 DBG: Plugin: Spiral2
2025-07-23 02:15:18.260 INF: lilv: 2025-07-23 02:15:17.220 DBG: Plugin: Srsly2
2025-07-23 02:15:18.260 INF: lilv: 2025-07-23 02:15:17.234 DBG: Plugin: StarChild
2025-07-23 02:15:18.260 INF: lilv: 2025-07-23 02:15:17.242 DBG: Plugin: SubsOnly
2025-07-23 02:15:18.261 INF: lilv: 2025-07-23 02:15:17.261 DBG: Plugin: ToTape5
2025-07-23 02:15:18.261 INF: lilv: 2025-07-23 02:15:17.282 DBG: Plugin: ToTape6
2025-07-23 02:15:18.261 INF: lilv: 2025-07-23 02:15:17.294 DBG: Plugin: TPDFDither
2025-07-23 02:15:18.261 INF: lilv: 2025-07-23 02:15:17.302 DBG: Plugin: TransDesk
2025-07-23 02:15:18.261 INF: lilv: 2025-07-23 02:15:17.312 DBG: Plugin: Tube
2025-07-23 02:15:18.261 INF: lilv: 2025-07-23 02:15:17.324 DBG: Plugin: Tube2
2025-07-23 02:15:18.262 INF: lilv: 2025-07-23 02:15:17.332 DBG: Plugin: TubeDesk
2025-07-23 02:15:18.262 INF: lilv: 2025-07-23 02:15:17.341 DBG: Plugin: Ultrasonic
2025-07-23 02:15:18.262 INF: lilv: 2025-07-23 02:15:17.349 DBG: Plugin: UltrasonicLite
2025-07-23 02:15:18.262 INF: lilv: 2025-07-23 02:15:17.357 DBG: Plugin: UltrasonicMed
2025-07-23 02:15:18.262 INF: lilv: 2025-07-23 02:15:17.372 DBG: Plugin: Verbity
2025-07-23 02:15:18.262 INF: lilv: 2025-07-23 02:15:17.389 DBG: Plugin: Vibrato
2025-07-23 02:15:18.263 INF: lilv: 2025-07-23 02:15:17.405 DBG: Plugin: XBandpass
2025-07-23 02:15:18.263 INF: lilv: 2025-07-23 02:15:17.420 DBG: Plugin: XLowpass
2025-07-23 02:15:18.263 INF: lilv: 2025-07-23 02:15:17.436 DBG: Plugin: XNotch
2025-07-23 02:15:18.263 INF: lilv: 2025-07-23 02:15:17.452 DBG: Plugin: XRegion
2025-07-23 02:15:18.263 INF: lilv: 2025-07-23 02:15:17.464 DBG: Plugin: ZOutputStage
2025-07-23 02:15:18.263 INF: lilv: new_state_from_model(): error: State 718b18 missing lv2:appliesTo property
2025-07-23 02:15:18.264 INF: lilv: 2025-07-23 02:15:17.497 DBG: Plugin Convolution Loader (https://mod.audio/plugins/ConvolutionLoader) has default state.
2025-07-23 02:15:18.264 INF: lilv: 2025-07-23 02:15:17.502 DBG: Plugin: Convolution Loader
2025-07-23 02:15:18.264 INF: lilv: 2025-07-23 02:15:17.518 DBG: Plugin: Marshall Bluesbreaker
2025-07-23 02:15:18.264 INF: lilv: 2025-07-23 02:15:17.545 DBG: Plugin: Dragonfly Early Reflections
2025-07-23 02:15:18.264 INF: lilv: 2025-07-23 02:15:17.576 DBG: Plugin: Dragonfly Plate Reverb
2025-07-23 02:15:18.264 INF: lilv: 2025-07-23 02:15:17.621 DBG: Plugin: Dragonfly Room Reverb
2025-07-23 02:15:18.265 INF: lilv: 2025-07-23 02:15:17.642 DBG: Plugin: ZaMaximX2
2025-07-23 02:15:18.265 INF: lilv: 2025-07-23 02:15:17.707 DBG: Plugin: ZaMultiComp
2025-07-23 02:15:18.265 INF: lilv: 2025-07-23 02:15:17.782 DBG: Plugin: ZaMultiCompX2
2025-07-23 02:15:18.265 INF: lilv: 2025-07-23 02:15:17.789 DBG: Plugin: ZamAutoSat
2025-07-23 02:15:18.265 INF: lilv: 2025-07-23 02:15:17.814 DBG: Plugin: ZamComp
2025-07-23 02:15:18.265 INF: lilv: 2025-07-23 02:15:17.844 DBG: Plugin: ZamCompX2
2025-07-23 02:15:18.266 INF: lilv: 2025-07-23 02:15:17.844 DBG: Plugin ZamCompX2 (urn:zamaudio:ZamCompX2) skipped. Not valid.
2025-07-23 02:15:18.266 INF: lilv: 2025-07-23 02:15:17.869 DBG: Plugin: ZamDelay
2025-07-23 02:15:18.266 INF: lilv: 2025-07-23 02:15:17.905 DBG: Plugin: ZamDynamicEQ
2025-07-23 02:15:18.266 INF: lilv: 2025-07-23 02:15:17.933 DBG: Plugin: ZamEQ2
2025-07-23 02:15:18.266 INF: lilv: 2025-07-23 02:15:17.992 DBG: Plugin: ZamGEQ31
2025-07-23 02:15:18.266 INF: lilv: 2025-07-23 02:15:18.016 DBG: Plugin: ZamGate
[... cut ...]
2025-07-23 02:22:00.841 DBG: Neural Amp Modeler: Staging model change: `/var/pipedal/audio_uploads/NeuralAmpModels/Neutron Quantum Speaker/Neutron QS-CABLUE-NDX/Quickstart/QS-CABLUE-NDX-01-BALANCED.nam`
2025-07-23 02:22:04.892 DBG: Neural Amp Modeler: Staging model change: `/var/pipedal/audio_uploads/NeuralAmpModels/Neutron Core Pedals/Drive/Neutron NC-DRV-BBRKR-NDX/NC-DRV-BBRKR-NDX-T06-D0.nam`
terminate called after throwing an instance of 'std::logic_errortrue
'
  what():  Timed out waiting for a Worker task to complete.
the terminal output..

rerdavies, 2025-07-23:
Huh. I actually need the `~/valgrind.output` file. But...!!! 

I think you have captured the message that gives it away.

    what(): Timed out waiting for a Worker task to complete.

That's my code. 

Probable fix coming your way.

rerdavies, 2025-07-23:
I have posted v1.4.82-Experimental, available here:

    https://github.com/rerdavies/pipedal/releases/tag/v1.4.82

It contains a fix that I sincerely hope will fix your problem. Let me know how it goes. 

The triggering condition is that there was a non-recoverable ALSA underrun while loading one of your NAM instances. And the crash occurred due a time out while trying to unload one of the PatchStorage NAM instances, as PiPedal restarted the realtime audio thread. (Not sure if that's actually useful information for you. Probably not).

I think you will still see underruns while changing NAM models (there's a LOT that goes on in the background load);   but hopefully, the plugins should be stable once loaded.

rerdavies, 2025-07-23:
(this is a fully optimized production build. So if it works, you can stick with it until the next generally available Pipedal release goes out.

StuSysMoo, 2025-07-23:
Ok i Tested it, and the issue remains... anything more i can give you?

StuSysMoo, 2025-07-23:
patch@Neutron-Pi:~ $
# Follow live output
sudo journalctl -fu pipedald
Jul 23 05:48:43 Neutron-Pi pipedald[927]: ALSA sequencer port  24:0 is already subscribed.
Jul 23 05:48:43 Neutron-Pi pipedald[927]: ALSA sequencer port  28:0 is already subscribed.
Jul 23 05:48:44 Neutron-Pi pipedald[927]: Audio started. ALSA, hw:UAC2, 32-bit integer little-endian, 48000, 32x4, in: 1/2, out: 2/2
Jul 23 05:48:44 Neutron-Pi pipedald[927]: Listening on mDns address Neutron-Pi:80
Jul 23 05:48:44 Neutron-Pi pipedald[927]: Listening on Wi-Fi address 192.168.3.68:80
Jul 23 05:48:44 Neutron-Pi pipedald[927]: DNS/SD service announced. (Neutron-Pi)
Jul 23 05:48:45 Neutron-Pi pipedald[927]: DNS/SD group established.
Jul 23 05:48:45 Neutron-Pi pipedald[927]: HotspotManager: Monitoring network status.
Jul 23 05:48:55 Neutron-Pi pipedald[927]: Listening on mDns address Neutron-Pi:80
Jul 23 05:48:55 Neutron-Pi pipedald[927]: Listening on Wi-Fi address 192.168.3.68:80
Jul 23 05:49:58 Neutron-Pi pipedald[927]: Web Socket Address: 192.168.3.68:80
Jul 23 05:49:58 Neutron-Pi pipedald[927]: WebSocketSession opened. ([::ffff:192.168.2.222]:60641)
Jul 23 05:50:37 Neutron-Pi authbind[927]: free(): corrupted unsorted chunks
Jul 23 05:50:37 Neutron-Pi systemd[1]: pipedald.service: Main process exited, code=killed, status=11/SEGV
Jul 23 05:50:37 Neutron-Pi systemd[1]: pipedald.service: Failed with result 'signal'.
Jul 23 05:50:37 Neutron-Pi systemd[1]: pipedald.service: Consumed 3min 36.034s CPU time.
Jul 23 05:50:42 Neutron-Pi systemd[1]: pipedald.service: Scheduled restart job, restart counter is at 1.
Jul 23 05:50:42 Neutron-Pi systemd[1]: Stopped pipedald.service - PiPedal Web Service.
Jul 23 05:50:42 Neutron-Pi systemd[1]: pipedald.service: Consumed 3min 36.034s CPU time.
Jul 23 05:50:42 Neutron-Pi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: Document root: /etc/pipedal/config Threads: 5
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: Locale: en_GB
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: Updater: Next update check at 2025-07-23 08:10:43
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: Scanning for LV2 Plugins
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Analyzer.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Analyzer.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Analyzer> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Equalizer12Band.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Equalizer12Band.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Equalizer12Band> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Equalizer30Band.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Equalizer30Band.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Equalizer30Band> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Filterclavier.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Filterclavier.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Filterclavier> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Fluidsynth.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Fluidsynth.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Fluidsynth> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Organ.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Organ.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Organ> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/Pitch.ttl (No such file or directory)
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_world_load_file(): error: Error loading file `file:///usr/lib/lv2/calf.lv2/Pitch.ttl'
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: lilv_plugin_get_name(): warning: Plugin <http://calf.sourceforge.net/plugins/Pitch> has no (mandatory) doap:name
Jul 23 05:50:42 Neutron-Pi pipedald[2589]: lilv: error: failed to open file /usr/lib/lv2/calf.lv2/presets-Reverb.ttl (No such file or directory)
[... cut ...]
Jul 23 05:50:43 Neutron-Pi pipedald[2589]: DNS/SD service announced. (Neutron-Pi)
Jul 23 05:50:44 Neutron-Pi pipedald[2589]: DNS/SD group established.
Jul 23 05:50:44 Neutron-Pi pipedald[2589]: HotspotManager: Monitoring network status.
Jul 23 05:50:49 Neutron-Pi pipedald[2589]: WebSocketSession opened. ([::ffff:192.168.2.222]:60713)
Jul 23 05:50:54 Neutron-Pi pipedald[2589]: Listening on mDns address Neutron-Pi:80
Jul 23 05:50:54 Neutron-Pi pipedald[2589]: Listening on Wi-Fi address 192.168.3.68:80

in case its useful..

StuSysMoo, 2025-07-23:
Hmmm... intersting.. that error occurred with the normal service running, but when i ran cd ~
valgrind  --log-file=valgrind.output /usr/sbin/pipedald /etc/pipedal/config /etc/pipedal/react -port 0.0.0.0:8080 -log-level debug and tried to trigger teh error, I couldnt make it happen for ages,  .. then it did.. so its definitely not triggering as easily for some reason running the valgrind command vs just normal service..
==3088== Memcheck, a memory error detector
==3088== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==3088== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==3088== Command: /usr/sbin/pipedald /etc/pipedal/config /etc/pipedal/react -port 0.0.0.0:8080 ->
==3088== Parent PID: 1830
==3088==
==3088==
==3088== Process terminating with default action of signal 1 (SIGHUP)
==3088==    at 0x4FF1C40: poll (poll.c:41)
==3088==    by 0x307B53: ??? (in /usr/sbin/pipedald)
==3088==    by 0x4CCE9DB: ??? (in /usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.30)
==3088==    by 0x4F9202F: start_thread (pthread_create.c:442)
==3088==    by 0x4FFBF5B: thread_start (clone.S:79)
==3088==
==3088== HEAP SUMMARY:
==3088==     in use at exit: 4,698,626,899 bytes in 276,222 blocks
==3088==   total heap usage: 59,568,884 allocs, 59,292,662 frees, 10,284,998,669 bytes allocated
==3088==
==3088== LEAK SUMMARY:
==3088==    definitely lost: 14,274 bytes in 254 blocks
==3088==    indirectly lost: 1,282,551,684 bytes in 4,871 blocks
==3088==      possibly lost: 212,060,436 bytes in 2,324 blocks
==3088==    still reachable: 3,204,000,505 bytes in 268,773 blocks
==3088==                       of which reachable via heuristic:
==3088==                         newarray           : 4,264 bytes in 1 blocks
==3088==         suppressed: 0 bytes in 0 blocks
==3088== Rerun with --leak-check=full to see details of leaked memory
==3088==
==3088== For lists of detected and suppressed errors, rerun with: -s
==3088== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

rerdavies, 2025-07-23:
=3088== HEAP SUMMARY:
==3088== in use at exit: 4,698,626,899 bytes in 276,222 blocks
==3088== total heap usage: 59,568,884 allocs, 59,292,662 frees, 10,284,998,669 bytes allocated

Doesn't look right. That's much higher than it should be.

The problem is definitely related to having to reset the realtime audio thread. And related to trying to deactivate the running plugins on the realtime thread. 

I need to give some thought to what's going on, and how best to deal with it. 

I have theories. But at this point, I'm going to have to call it a night and go at it with fresh eyes tomorrow. 

I do get the general gist of the problem. And I can think of things to do to mitigate the issues. 

More tomorrow.

rerdavies, 2025-07-23:
Status update: I have theories as to what's going on, and code in progress to address those theories.  A new release is on the way, but the changes are not entirely trivial.  I expect to issue a new exerimental build sometime later today.

My current read: a perfect storm triggered by hard ALSA underruns. PiPedal restarting the audio thread, and in the process, unnecesarily  reloading all the plugins from scratch. Which generates more underruns, which possibly causes timeouts on the connection between the server and the client because it's starved for CPU time.

And SOMEBODY is leaking memory at a furious rate. 100MB per restart! I'm debugging that right now. It's possible that the PatchStorage NAM is trying to invoke LV2 Worker functions after being deactivated (most particularly the  Worker tasks that free the memory occupied by their NAM models..  

One thing that's clear at this point: there WILL be a buffering version of the TooB NAM plugin to address some of these issues properly. In the meantime, the question is how to get PIpedal to survive the odd things that the PatchStorage plugin is doing.

As "one more problems go", this is a good one. :-)

## Source code for the NAM plugin.

You mentioned that you had actually managed to compile the MOD NAM plugin. One of the challenges I'm currently facing is not having source code for the PatchStorage plugin. And figuring out how to apply the DIFFS in the actual MOD source tree iin order to get the actual source for the plugin is not obvious. I can build Mike's NAM without problems. But not the Patchstorage NAM plugin.   If you have that project handy, is there any way you could extract the sources they use to compile the plugin from the VM(?) they use to build the plugins?  The procedure they seem to use: they download an outdated version of Mike's NAM, and then apply a DIFF file to Mike's ssources in order to patch in the buffering code.  The entire directory of name-specific source, and all subdirectories (if any). The directory in question will contain the following files (and perhaps others):

```
nam_lv2.cpp
nam_plugin.cpp
nam_plugin.h
```

There's clearly a big problem in their Lv2 deactivate implementation, and it would be nice to just KNOW what they're doing rather than having to speculate.

Extra difficult, because I have had previous interactions with FalkTX (author of the MOD software, and also probably the buffering code), and he and I have divergent opinions on how LV2 hosts are supposed to behave. Interactions that were, as a matter of fact, directly related to changes to Mike's original NAM plugin. PiPedal was actually revised to mimic a couple of the MOD host's interpretations of the LV2 spec.  So getting some idea of what HE thinks the shutdown behaviour should be would be extremely useful at this point.  .oO( of which one of the particular interpretations is going to introduce an extra buffer delay. sigh another reason why the buffer processing shouldn't be done on the LV2 Worker thread). And probably challenging because he knows that the plugin doesn't really work on MOD Dwarf anyway, so he probably doesn't really care. :-/

## Does your Pi have all updates applied.

Also a curious thing... the problems are occuring in code that is triggered when ALSA needs a hard restart after underruning. Normally, there's a short sequence of code that clears the underrun condition when buffers undderun. And sometimes ALSA decides that it wants the ALSA devices closed and re-opened in order to restart. 

I am unable to trigger the hard-reset condition on my current version of Raspberry Pi OS. It certainly used to happen on previous versions of Raspberry Pi OS, but it doesn't happen anymore, no matter how I mistreat the ALSA streams. Which makes me think that there's a recent update that prevents the hard-reset-required condition altogether (it does seem rather unnecessary and arbitrary). There are SUBSTANTIAL updates to the system audio code (both ALSA and pipewire). Which makes me wonder: 

Do you have all current raspberry pi apt updates applied?  

Whether you do or not doesn't actually matter. I run into the full restart path when audio devices are reselcted in the settings dialog. And if there a memory leaks (there are), I still need to fix that problem. But I do suspect that updating your system might cure the immediate symptoms you're dealing with.  But it would give me reassurances that my testing on a Pi 4 is going to replicate your experience on a Pi 5.

StuSysMoo, 2025-07-23:
I am using patchbox OS with all updates applied (for the Preempt RT Kernel).  I managed to find teh correct commit for the mod patch and grab a cached version yes, I can write up a guide on getting it to work if thats useful?

StuSysMoo, 2025-07-23:
git init
git remote add origin https://github.com/moddevices/neural-amp-modeler-lv2.git
git fetch --depth=1 origin bae8b0a6275a42ebbfe9c9d957512419700dc797
git checkout FETCH_HEAD

that should bag oyu the cached commit needed for:

curl -O https://raw.githubusercontent.com/mod-audio/mod-plugin-builder/master/plugins/package/neural-amp-modeler-lv2/01_mono-buffered-optimizations.patch

then:
patch -p1 < 01_mono-buffered-optimizations.patch

the patch works only against this: bae8b0a6275a42ebbfe9c9d957512419700dc797

rerdavies, 2025-07-23:
Thank you for the build instructions. It was that patch step that I was missing.

In the meantime...

Fresh new build. Fixes in this build: reworking of the audio thread restart mechanism; a fix for the memory leak. I have high hopes for this one. 

    https://github.com/rerdavies/pipedal/releases/tag/v1.4.83

Keep me informed please.

StuSysMoo, 2025-07-24:
i made a GDB backtrace, maybe more useful... neutron@Neutron-Pi:~ $ sudo gdb /usr/sbin/pipedald 7634
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/sbin/pipedald...
(No debugging symbols found in /usr/sbin/pipedald)
Attaching to program: /usr/sbin/pipedald, process 7634
[New LWP 7637]
[New LWP 7642]
[New LWP 7644]
[New LWP 7645]
[New LWP 7646]
[New LWP 7647]
[New LWP 7648]
[New LWP 7649]
[New LWP 7650]
[New LWP 7651]
[New LWP 7652]
[New LWP 7653]
[New LWP 7654]
[New LWP 7655]
[New LWP 7661]
[New LWP 7662]
[New LWP 7663]
[New LWP 7664]
[New LWP 7665]
[New LWP 7668]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
0x00007fff4aabb6d8 in __GI___sigtimedwait (set=set@entry=0x7fffefdc12c0, info=info@entry=0x7fffefdc1018, timeout=timeout@entry=0x0) at ../sysdeps/unix/sysv/linux/sigtimedwait.c:31
31      ../sysdeps/unix/sysv/linux/sigtimedwait.c: No such file or directory.
(gdb) continue
Continuing.

Thread 8 "ppdl_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff2ffff0c0 (LWP 7648)]
0x00007fff44582f34 in ?? () from /usr/lib/lv2/neural_amp_modeler.lv2/neural_amp_modeler.so
(gdb) bt full
#0  0x00007fff44582f34 in ?? () from /usr/lib/lv2/neural_amp_modeler.lv2/neural_amp_modeler.so
No symbol table info available.
#1  0x00007fff4459f10c in ?? () from /usr/lib/lv2/neural_amp_modeler.lv2/neural_amp_modeler.so
No symbol table info available.
#2  0x00007fff4ab02030 in start_thread (arg=0x0) at ./nptl/pthread_create.c:442
        ret = <optimized out>
        pd = 0x0
        out = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140733998690496, 140737217561808, 140737217561398, 0, 8448384, 140737217561399, 140733990240256, 140733998691544, 140733990240256, 140733998692352, 140733998688592,
                13916844001026308861, 0, 13916843999470881793, 4676829883349860352, 0, 0, 0, 0, 0, 0, 0}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#3  0x00007fff4ab6bf5c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
No locals.
(gdb)

StuSysMoo, 2025-07-24:
I have been doing some legwork of my own and its becoming very clear this bug is in the mod plugin, or more specifically in teh changes introduced by the mod patch... here is a summary of my current understanding of the situation, forgive the sterile nature of this output its really just me trying to take stock after a couople of days of grinding away at it. Originally i was trying to fix the mod patch so it doesnt cause the nasty bug, but it may be better for everyone if your intention is to add the buffered switch to the Toob plugin to abandon the mod stuff entirely as it has no path without extreme pain to patching an up-to-date NAM version, and mod themselves seem to , as you say, have abandoned it anyway.. we have the patch, we know what they did.. its just a case of doing that but without the bug.. Your work would bring this feature into the main NAM pipe as a maintained solution and allow it to benefit from future upgrades and innovation.. I dont want to flog this dead mod horse any longer than is useful, but I would hope my findings can help you at least match the huge efficiency gains they achieved with their patch wihtout the crashing.. 

Background:

MOD Devices created a heavily optimized version of the Neural Amp Modeler LV2 plugin for their platform, including a “buffered” patch that leverages SIMD instructions for performance gains—especially in the final output copy step within the core convolution network code (convnet.cpp).

The MOD fork is currently unmaintained. The PiPedal project (and your ToobNeuralAmpModeler plugin) are likely the best places for further development and user benefit.

The Issue:

When applying the MOD “buffered” patch to Neural Amp Modeler LV2 and running it on ARM platforms (e.g., Pi), the plugin crashes with a segmentation fault.

GDB traces show the crash occurs at a SIMD load instruction (e.g., ldr qXX, [x0]) inside the output buffer copy in convnet.cpp—due to an unaligned pointer.

The patch assumes the destination buffer is always 16-byte aligned, which is true on MOD hardware but not always true on Pi or other ARM systems.

Why It Matters:

The “buffered” logic offers significant performance improvements—making real-time amp modeling feasible even on low-power ARM devices.

However, the current implementation is not ARM-safe and fails unless strict alignment guarantees are met.

This issue is a barrier to using or integrating these optimizations into PiPedal/ToobNeuralAmpModeler, and resolving it would benefit the entire LV2/Linux audio community (especially as MOD Devices is no longer maintaining their code).

What We Know:

The crash is not due to algorithmic error, but to hardware alignment requirements for SIMD/NEON instructions on ARM.

If the output buffer copy is made alignment-aware (checking and handling unaligned pointers), the plugin should run safely and efficiently on all platforms.

The MOD patch otherwise seems to deliver as promised—excellent DSP performance, with a single point of fragility on non-MOD ARM hardware.

Next Steps / Potential:

Solving this one issue (buffer alignment in the output copy loop) could unlock the performance benefits of the MOD Devices approach for PiPedal and other open platforms.

Your interest in bundling an improved, buffered Neural Amp Modeler as part of PiPedal is exciting, and would fill a gap left by MOD’s withdrawal.

StuSysMoo, 2025-07-24:
[neutron-buffered-conceptual.patch](https://github.com/user-attachments/files/21418570/neutron-buffered-conceptual.patch) here is a WIP conceptual patch file that has the goal of fixing the issue in MOD's patch file, so it could be applied to build a plugin that doesnt have the misalignment problem.. just for reference

rerdavies, 2025-07-24:
<exhaling> Huge sigh of relief. I'm glad that's over, even it if it ends badly.

Although, as a side note, I'm pretty sure Arm 8 neon allows 4-byte alignment on SIMD memory arguments. There's a very small penalty for non-16-byte-aligned access, but I'm pretty sure NAM (and its dependent libraries) use best-possible-alignment memory allocators. Which would suggest actual memory corruption (a pointer with a non-mod-4  low-order byte) rather than a problem with memory allocators.  

Nevertheless, I agree with your point about supportability. Background processing is actually fairly straightforward to implement in Toob NAM. It will definitely take less time to implement the feature in TooB NAM than we have spent on the problem so far. 

Expect something in the next couple of days.

rerdavies, 2025-07-24:
Correction: Arm 8 neon allows unaligned accesses, but a common optimization is to provide alignment hints to the neon instructions which reduce the throughput of neon instructions by one CPU cycle. Hinted neon instructions will fault if fed unaligned memory. And .. from the snippet ... Nam CORE does use hinting. So for our purposes, 16-byte alignment is required.

rerdavies, 2025-07-26:
Looking good. I'm just running through final testing right now. 5 instances of TooB NAM on a Pi 4. We'll see what you get on Pi 5. I'll have a preview build for you in about half an hour.

Neutron-Stu, 2025-07-26:
cool! i actually managed to build a fully Pi-5 optimized "standard" Mike-O
NAM yesterday (without buffered patch) and got up to 8 (at 96% CPU)
instances of THAT on pi5, however it seems to have some issues when you
load "none" as the nam, throws an error in the gui. non-issue if you are
progressing Toob though.. more my curiosity on the current state of Mike's
repo and what performance level the base code has reached without buffered
improvements..
  -DCMAKE_BUILD_TYPE=Release \
  -DCMAKE_C_FLAGS="-mcpu=cortex-a76 -march=armv8.2-a -O3 -flto" \
  -DCMAKE_CXX_FLAGS="-mcpu=cortex-a76 -march=armv8.2-a -O3 -flto"
these were the flags i used.. seems quite a lot faster than the old one.

I have turned my attention more towards prep of the Neutron Prime concept
device and specifically the interface for the touch screen, building
something on Pi-OS lite with Plasma-Mobile DE that looks and works in a
very "phone-like" manner highly in-line with your PiPedal ethos.. I will
also need to build my own fully RT kernel too, so lots to get on with.. Let
me know if you want anything tested, I have some ideas also around the
other elements in Toob Nam (eq, gate, etc) that a couple of tweaks could
really increase the value of those extra controls.. but let's get it
working first, then get it really smoking.. Stability and low latency are
the current order of the day. Thanks for keeping me up to date, if there is
anything I can help with let me know...

rerdavies, 2025-07-26:
Interesting. It would make sense that one would get a big performance boost
out of using A76 compiler options. I shall have to see what he's done. And
push a change to Mike's repo for the "NONE" selection. That's a very easy
fix. (We have collaborated before). I suppose I could conditionally install
an A72 and A76 version of pipedal.

(I also have an active collaboration underway to get PiPedal to download
NAM DIRECTLY from TONE3000.com. An in-pipedal dialog/plugin browser that
would allow you to upload to the PiPedal server from TONE3000 in one click.
Just received a prototype that I urgently need to provide turn-around on.
So between this and that... I am giving your work priority for the moment
though.).

Re running all on one device. I'm not sure what you're currently doing, but
be aware that GPU activity can have undesirable effects on real-time audio
stability. (One of the reasons why I really like the remote phone UI that
PiPedal has). Network activity is perfectly fine (on a Pi); disk/storage
activity - not quite so much, but nowhere near as dire as GPU activity. '

I am definitely interested in feedback on usability, and extra effects.
We'll close the loop on that after all of this is done. Also interested in
your pedal color palette. It occurs to me that I could put "Icon Color" in
the new editor for plugin labels (click on the plugin title). So I'd like
to chat about that before we put everything to bed.

Your build is cooked.  Going to upload it now.

Neutron-Stu, 2025-07-26:
Be careful going too far down the tone3000 route.. the sounds on there are,
for the most part, not so great and it's not the best representation of
NAM' true potential.. It's difficult I guess as there will be lots of
community pressure there, and a perceived value in the sheer volume of NAM
files on the platform.. but honestly the idea of it being integrated into
the actual plugin makes me shudder a little.. I have some very strong
opinions about it. I think at some point it would be really cool to have a
proper chat about your vision for PiPedal in future, and over my 30 or so
years as a pro guitarist, multi-platinum selling producer, recording and
mix engineer I have more than a few ideas about things you could add to
make what you have truly "pro grade". I have no right to ask you to listen
to me over anyone else though at the end of the day, but I am happy to
offer any insights and experience I can to help you because I genuinely
like what you have done and think it's truly a special piece of work that,
regardless of the direction you go I will certainly find many uses for..
For my part I would rather work in collaboration than try to fork off and
customize the project. I believe once I can show you some of what I am
planning to do with PiPedal you will see the logic in many of my requests
anyway.. but that's for later..

rerdavies, 2025-07-26:
To be honest, the immediate challenge it solves for me is providing users with an easy way to install models so that they can instantly realize how incredibly good TooB NAM is without having to go through a cumbersome download/upload procedure.

I've tried to source sample models for PiPedal. But it is surprisingly difficult to do, if I want to be meticulous about how I handle the Intellectual Property issues. (And I do. As should every professional developer).

TooBML  has a similar problem . It does bundle starter sets of ancient models, probably trace their heritage to  Jatin Chohowdry's PhD thesis. They are not of good quality though. And mask the fact that TooB ML with good models is just as good as TooB NAM, with considerably less compute cost. 

What I need: models that can be bundled with PiPedal that have a clear and unambiguous open-source-compatible license associated with them (probably CC-BY of some flavor, since they are data files).  I'm a professional programmer so I take intellectual property rights seriously. So I can't bundle files that do not have clear licensing conditions attached.  (For model files, I dont' see a problem with CC BY NC ND, or other semi-restrictive open-source license. Pure CC BY seems like too much to hope for). If PiPedal ever 

Just two or three so that users can effortless try out TooB NAM and be instantly amazed.

Would you be interested in a possible co-promotion? :-) A couple of isolated models from your amp packs in exchange for direct links to your site from Toob/Pipedal help and documentation files?  A chance to reveal to the world how good your models are? 

:-) 

That didn't actually start out as a pitch. But, by the time I got to the end, it seemed like an awfully  good idea. 

I definitely think we should get together and have a discussion about this, and other various bits and pieces that have accumulated throughout this support  case. (Like your Boss pedal colors).

rerdavies, 2025-07-27:
And, because I am burningly curious... 

Here's a build with the TooB plugins optimized for A76... Exactly the same as the 1.4.85 build, but compiled for a Cortex A76.

https://github.com/rerdavies/pipedal/releases/download/v1.4.85/pipedal_1.4.85_arm64-a76.deb

It does indeed look like the A76 has new instructions that are virtually purpose-built for running neural networks. So substantial performance improvements are definitely a possibility.

If you feel so inclined, I would be very interested to know if there's a significant difference in performance.  It's not that difficult to provide A72 and A76 versions of TooB NAM in the PiPedal debian package.(Although building them will be hell).

Neutron-Stu, 2025-07-27:
I did some testing of this version, and of the Toob Nam too, and here is
what I found:

I got up to 15 instances of the Toob NAM plugin at 128/3 buffers on the
current state of the Neutron Prime Box (that's the highest I would regard
as practical) before xruns started.. pretty impressive stuff. I tested the
old MOD buffered LV2 and it managed an astonishing 18! The Toob Plugin i
noticed suffers from an issue when changing the NAM with many instances
running, it will seemingly kill the audio engine, quite predictably when
you change any of the NAMs once you have a lot of instances running, only a
service restart will bring the audio back.. I did notice another thing that
could be an issue later when we start to look at the application of this
system in a live environment, and that is when you change the NAM in a
slot, like load a snapshot etc, there is a quite noticeable dropout in
audio as it loads the new one, around 0.3 seconds.. But this also seems to
kind of gang up when there are a few instances and make the
loading/changing of NAMs in the Toob plugin feel very sluggish.. Make it
work then make it work fast i suppose :)  This does not occur on any of the
Mike O based ones, it's absolutely instant and seamless when switching.
That is highly desirable when using things live and changing snapshots,
it's super clean and fast, but the Toob needs some work there to get it to
the same state I think.. There are a few things regarding live use that I
would like to work on once we get past this initial work on the NAM part,
some of them will be very simple to implement, others may take more
experimenting.. All in though, as our starter for 10 points.. pretty
amazing gains in performance!

rerdavies, 2025-07-27:
## Performance of the A76-optimized TooB NAM. 

Could you clarify that please. There's now four combinations to deal with:

-  A76 build, with threading.
-  A76 build, without  threading.-
-  A72 build, with threading.
- - A72 build without threading.

18 is excellent. But for which combination?

## Buffer configuration. 

There is a problem with your buffer configuration.  128x buffer sizes are going to introduce too much latency to make your 18 NAM instances usable. Each instance of a buffered/threaded NAM introduces one additional buffer delay in the signal chain. So if you were to configure 18 threaded NAM instances, with a buffer size of 128, you're looking at end-to-end latency of ... 128*(18+3)/48000 =0.48 .... 48 milliseconds of delay in the signal, just counting buffer delays which is no longer in a playable range.  The general consensus is that total signal delay has to be under 10ms, but that less is good.. 

I would actually be more interested in how many you can run in a more practical buffer configuration. My preferred buffer configuration, currently, is 32x4. Although 64x3 and 64x4 may be better in this situation, with really high CPU loads.  I don't think that 32x buffer configurations actually gets you into playable range. But you'd probably get usuable results with 8 NAMs. (I'll leave you to do the calculation yourself). 

It may well be the case that you require a 64x buffer to get everything to run stably. But it is my current belief that 32x4 is actually better than 64x3 in terms of latency and stability.

## Dropouts in TooB NAM while loading. 

This is actually by design, and borrows an idea from MOD Nam, which Mike's NAM does not implement. 

The issue being addressed: NAM models need to be "pre-warmed". When freshly loaded, they have significant DC offsets in their output, and it takes up to 1/4 second for the internal state of the model to reach equilibrium, and become usable. (Not that different from a Tube Amp turning on, really). 

There's three different ways to deal with the pre-warming issue. 

(1) Pre-warm the models on the background thread immediately after loading them. NAM Core does this by default, although it does provide an option to bypass pre-warming altogether. And Mike's Nam does this.  Doing so significantly impairs performance of the realtime thread,  Probably because the NAM code is very memory-intensive (more so than it is compute intensive), and because each NAM instance accesses the L2 cache in a similar pattern. So running extra instances while loading is very like to cause underruns. Why wouldn't it? You have temporarily double the number of NAM instances running on background threads. So it tends to cause overruns. And the switches between models are not de-zipped. There's a hard switch, which means that there's a significant risk that there will be a hard pop in the signal. 

(2) Cheat on the pre-warming, and warm the models much less than they should be warmed. Previous versions of Toob Nam used this strategy. You get a mild "pop". Sounds awfully like a tube amp turning on. Not a full-on spike in the signal; but a pop nonetheless.

(3) Pre-warm the model on the realtime thread. MOD Nam does this (and I borrowed the idea). The models are not pre-warmed on the background thread, but are instead pre-warmed on the realtime audio thread. These means that they have to be muted for 1/4 second. Pros: a completely pop-free transition when loading models. The previous model is faded out on the real-time thread while the new model is loading. When the new model comes back to the realtime thread, it is run, but muted for 1/4 seocnd, and faded in without a pop once ready. (Or muted and faded in on the buffer thread, as the case may be. Also very nice because one entire frame of audio has to be inserted or removed when switching between threaded and un-threaded operation. And the extra frame will now be all zeros because of the muting.

Toob NAM now uses the third approach. The behaviour changes, but I think it's for the better. If you are actually playing when the model changes, your output will duck as the model changes. But if you were actually playing, on a relatively clean amp, you would hear pops and clicks as the model changed. No more pops and clicks. 

And if you are not playing, you hear nothing but blissful pop-free underrun-free silence. Which is going to be the actual case when playing live. I don't think anyone realistically expects to switch back and forth between models without expecting some kind of discontinuity in the signal.

So I think the behavior is different, but better. It would require a pretty strong argument to make me think that this isn't the right thing to do.

Neutron-Stu, 2025-07-27:
Sorry I should have explained, this was not a playability test, more a way
to ensure the CPU can work to its full capacity regardless of
the performance of the audio interface, as you know they are not all
created equal in that regard.. when you get above 128s that will become a
non-factor, whereas any testing we do under 128 will "bake in" the
performance of the card in the results.. meaning on my UAC-2 I may be able
to run 10 instances at 32/3, but for another card that may not also be
true.. Do you see the logic? That may be fine as long as we only ever test
with the exact same components.. but that may not always be possible. If
you want to fix a low buffer yardstick for testing, we just need to be
aware that it comes with baggage. My goal is to understand the raw
computational status of the load on the host rather than evaluate
playability at this stage. I am also a big fan of 32/4, but you need a
class compliant UAC2 capable interface to get into the 80+% cpu load
without issues at those buffer settings, and those who have the ideal cards
will be the low 10% of the user base unfortunately, so we should understand
the raw load first and then i have about 10 different interfaces I can run
the same test against at lower latencies to see how the lan lies there. I
don't mind if you  want to fix a specific buffer size as the global
standard for all testing, but I am conscious there will be many widely used
cards that cannot break even 5 instances at 32/4 without issues... All of
those results were on the A76 build, with threading enabled-vs Mod-NAM with
buffered enabled. I can do a series of tests with any specific combination
of factors that are useful to you, just let me know what you would like me
to rig up and I can work through them...  I am not sugar coating the
patch-gap thing because I know the user base, and I know their pet peeves..
This will come up if not addressed at some point i can pretty much
guarantee it. Patch transition (particularly in snapshot changes) is an
area often used to differentiate the "pro" form the "bedroom". I will hold
to the idea that PiPedal can and should be the gold standard and you can
decide if that information is useful or not, but I see no reason to sugar
coat things as long as you are happy to accept that it's all for the best
possible reasons.. I can certainly accept that some things are simply not
possible or practical, and am not looking for hills to die on, just expect
me to aspire towards excellence, all with the best of intentions, I meant
no disrespect in my comments, so apologies if that was not welcome
commentary.I am at your service to perform all manner of testing, just say
the word.

Neutron-Stu, 2025-07-27:
Incidentally, the buffered config is not cumulative like that in reality..
again i will do some precise measurements on exactly what it does do, but i
can say for certain, there is not 48ms of added latency in buffered mode
with any number of added instances.. it is more nuanced than a simple
addition of buffers i think.. All will be revealed once I get my lab coat
on and start to measure it all more precisely, I was more concerned gettin
over the first hurdle of all the pieces working as intended.. but it seems
we are nearly there now :)

Neutron-Stu, 2025-07-27:
Sorry i didn't answer your last question: 18 instances was: Mod-NAM (the
old Mod-patched Mike-O plugin available here:
https://patchstorage.com/neural-amp-modeler/ with 128/3 buffered on. up to
18 instances i had 0 xruns over about 5 minutes. 19 broke the camel...

rerdavies, 2025-07-27:
I would be exceedingly grateful if you could confirm that the A-76-optimized 1.4.85 build does what it's supposed to with threading enabled.  That way I can close off this if this bug.  I think it' s otherwise complete, once you confirm that the new TooB plugins address the issue. (There are two important bug fixes to PiPedal, that were uncovered by testing with Pastchstorage NAM, so you know. This has been enormously worthwhile).  

If you see the same sort of improvements with the A76 optimizations on the Toob plugin, I'll work out a way to provide A76-optimized versions of at least the TooB plugins.  Ducking issue aside, the new plugin has so far passed all testing at my end. We can continue that discussion in an email thread, if you like. 

There's lot's more I'd like to chat about, though. Perhaps we could switch to email from here. My address: rerdavies@gmail.com 

Re the consumer- versus pro-level USB adapters. I wince in exactly the same way you wince about the TONE3000 models. Some of the entry-level consumer devices have horrible signal to noise ratios ... in at least one case, only 10-bits of signal to noise. This is a disaster when running NAM models, because, by the time the input signal has been overdriven, you're only getting 12- or 18db of signal above noise floor. Which sounds horrible. In no way superior to previous generation of amp modelling. If on the other hand, you use a pro-level USB adapter with 120dB SN/R, it's like a veil comes off. Instant clarity and definition. Models that truly sound like their originals. And even feel like their originals when you play them (which I find the most extraordinary part of it all. I intensely dislike the feel of amp models in Helix-class devices, as compared to playing on a real tube amp)). If essential problem: if you don't have signal quality going in, you don't realize the full potential of NAM. (Who said that?).  And obviously if you don't have models that were produced with comparably meticulous audio engineering, on comparably good audio adapters, the same problem applies.

It's not ACTUALLY clear to me that small buffer sizes are a problem with consumer-level audio adapters. All of these devices are using the same drivers, since Linux supports only class-compliant USB interfaces. And - as long as they're using USB 1.1 Audio (pretty sure that USB 1.0 Audio devices are completely relict at this point) they should not have a problem with bandwidth or throughput., and only modest problems with hardware latency. Although I have dealt with one support case  with a consumer-level USB device that has a minimum buffer size of 512. So outliers do exist. My current belief is that they are rare, and old devices.  If you have evidence to the contrary with your pool of test devices, I'd love to hear about it. And the Behringer device should actually be a very good device. If you are using it as proxy for consumer-level devices, it may not be the best choice. Go for something horrible, like M-AUDIO, which will probably use the same chipset that all the rest of the entry-level consumer devices use. (Burr-Brown/TI).

That being said, I am in the midst of taking a pull request to support in-only and out-only USB adapters, all of which I expect to be horrible devices. The pull request is a bit rocky, but it will clear eventually.  And have serious concerns that it's going to open me up to a slew of support cases for devices that I have no real desire to support. Particularly worried about the family of Mac-targeted in-only USB adapters, and in-only "Guitar cable" USB devices, which all look a bit disrepuatble. Anyway. Too late now. 

Reply to my email please:

rerdavies@gmail.com

rerdavies, 2025-07-27:
And concerns about senstive issues... Not at all. Quite the opposite. Your credentials get you a seat at the head of the table when it comes to input on PiPedal. I am extremely interested in what you have to say. 

Patch switching: Pipedal does some fairly extraordinary things, but admittedly isn't 100% there yet compared to a ROM-based stomp box. We can also discuss that offline.

rerdavies, 2025-07-27:
:-) I wrote it. There is a one buffer delay. Write to the background
thread, pick up the results in the next sample frame. Same for the Mod NAM
plugin. Delaying for one buffer is precisely the premise on which the trick
is based.

closed 2025-08-12 by rerdavies