rerdavies/pipedal #348: Documentation: Disable lightdm so it won't make audio stop every 10 seconds

avanzzzi, 2025-07-23:
Here is a suggestion to include in the tips or FAQ section:

There was a moment, I believe after a system update, when my two audio interfaces started to crash every 5 or 10 seconds, regardless of the buffer size used. Upon checking the system, I noticed that the lightdm service was enabled and running, even though I hadn't installed anything GUI related.

So I disabled the service with:
```sudo systemctl disable lightdm
sudo systemctl stop lightdm```

And everything started working fine again. 
I believe that this can help others who go through similar problems.

rerdavies, 2025-07-23:
So, lightdm is --essentially-- a login screen manager. I'm not entirely sure that's the right way to disable it.  Did you switch from auto-login to non-auto login?

And,... which OS are you using? 

Interesting that it kills audio, though. That is FAQ worth. (And it is about time I added a FAQ Page! :-) )

avanzzzi, 2025-07-23:
I'm using raspberry pi OS bookworm, setup was done entirely headless, not even VNC enabled.

Yeah, on a more "common" system, lightdm service would be triggered by systemd as a target for multi-user system. I have to check if it's enabled in my multi-user.wants sctructure yet. Stop and disable is a quick dirt way to do it but, you are right, maybe I should look further for an apropriate way to write this down on a FAQ.

Another thing I might look into is that if it's still active even after login as it does on PC installations.

I`ll setup the system and pipedal from scratch again to figure out what triggered lightdm in the first place.

Thank you!

rerdavies, 2025-07-23:
That's disturbing news. :-/ Headless Raspberry Pi OS is the *last* place I would expect lightdm to show up. 

I appreciate your volunteering to take the time to research follow-on issues.  I'm patching in a FAQ page right now, with a spot reserved for whatever you find out. It does indeed look like lightdm causes problems with audio on non-raspberry-pi OSs. 

A tip from an AI: there's a setting to disable the "log on sound" somewhere. If that's the root cause of the issue, the fix going forward may be for the pipedal debian installer to turn that off, since it has root privileges during the package install procedure. 

But it probably needs a FAQ entry anyway. 

I'll wait to see what you find out.

rerdavies, 2025-07-25:
Interesting that wireplumber is getting involved.  Did you install wireplumber? That would  be consonant with reported problems on other platforms wrt/ the "Login sound" that lightdm plays. And chilling that a bluetooth media endpoint fails to start up at about the  time bad things happen.

```
Jul 24 18:19:56 raspberrypi dbus-daemon[13335]: [session uid=105 pid=13335] Activating via systemd: service name='org.freedesktop.impl.portal.PermissionStore' unit='xdg-permission-store.service' requested by ':1.8' (uid=105 pid=13332 comm="/usr/bin/wireplumber")
```

For what it's worth, I'm halfway through building a FAQ page, and I'm short of Frequently Asked Questsions (I have a couple of good ones). So this issue may end up on the FAQ page anyway, even if it isn't Frequently Asked.