rerdavies/pipedal #355: [BUG] Unable to update though UI

FoolHen, 2025-07-25:
I tried updating through the UI but it never updates, says downloading and then installing and then the dialog disappears and nothing happens. Sometimes the app disconnects from the pipedal after the installing dialog but right after the hotspot is available again and when opening it is back as it was before.

Trying to update from v1.4.76-Release to v1.4.79-Beta. 

Linux 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt2 (2024-10-01) aarch64 GNU/Linux

FoolHen, 2025-07-25:
```
Jul 25 18:42:53 fulpi systemd[1]: pipedald.service: Failed with result 'signal'.
Jul 25 18:42:53 fulpi systemd[1]: Stopped pipedald.service - PiPedal Web Service.
Jul 25 18:42:53 fulpi systemd[1]: pipedald.service: Consumed 43.607s CPU time.
Jul 25 18:42:56 fulpi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Jul 25 18:42:56 fulpi pipedald[1932]: Document root: /etc/pipedal/config Threads: 5
Jul 25 18:42:56 fulpi pipedald[1932]: Locale: en_GB
Jul 25 18:42:56 fulpi pipedald[1932]: Error loading library: libicui18n.so: cannot open shared object file: No such file or directory
Jul 25 18:42:56 fulpi pipedald[1932]: Updater: Next update check at 2025-07-26 02:36:21
Jul 25 18:42:56 fulpi pipedald[1932]: Scanning for LV2 Plugins
Jul 25 18:42:57 fulpi systemd[1]: Started pipedald.service - PiPedal Web Service.
Jul 25 18:42:57 fulpi pipedald[1932]: Found ALSA device hw:sndrpihifiberry.
Jul 25 18:42:57 fulpi pipedald[1932]: Audio started. ALSA, hw:sndrpihifiberry, 32-bit integer little-endian, 44100, 32x4, in: 1/2, out: 1/2
Jul 25 18:42:57 fulpi pipedald[1932]: Listening on mDns address fulpi:80
Jul 25 18:42:57 fulpi pipedald[1932]: Listening on Wi-Fi address 192.168.2.229:80
Jul 25 18:42:57 fulpi pipedald[1932]: DNS/SD service announced. (FulPedal v1)
Jul 25 18:42:58 fulpi pipedald[1932]: DNS/SD group established.
Jul 25 18:42:58 fulpi pipedald[1932]: HotspotManager: Monitoring network status.
Jul 25 18:43:01 fulpi pipedald[1932]: WebSocketSession opened. ([::ffff:192.168.2.147]:53160)
```

extremesecrecy, 2025-07-25:
I think that "library_handle = dlopen("libicui18n.so", RTLD_LAZY);" is using ibicu-dev. You can try to install it:

sudo apt update
sudo apt install libicu-dev

Then reboot and try the upgrade.

rerdavies, 2025-07-26:
The missing libicu-dev dependency is actually fine. PiPedal falls back to sorting using US-ASCII format if the library is missing. With the library present,  various lists of things are displayed in full locale-dependent sort order.  But you can install it if you like.

rerdavies, 2025-07-26:
For what it's worth, the libicu-ev dependency has been added as an
installer dependency in the debian package in versions of PiPedal going
forward. Using US/C sort order is an unpardonable sin. I had incorrectly
assumed that libicu was a pre-installed package. However, te
missing library is completely harmless at this point. Except for mis-sorted
lists on Swedish devices. When plugin names start with accented
characters...., which doesn't happen that often... :-/

FoolHen, 2025-07-27:
> What I was actually interested in was system logs through an attempt to update. You said that the in-app update was failing (or maybe I misunderstood).

I'm pretty sure those were the logs. I guess the update process didn't really start? In any case, I just tried again today and it updated on the first try. Quite strange, since I tried multiple times before (probably like 15 or 20 times) in different days too, and never worked. 

I can roll back and try again if you want. But I think there was nothing else in the logs.

rerdavies, 2025-07-27:
Hmm  Lets leave it at that for now. I don't really have any theories. And I
haven't been prompt enough that we might have been  able to catch logs from
when it did happen.

closed 2025-08-13 by rerdavies