rerdavies/pipedal #363: [BUG] output level of stock presets

jardous, 2025-07-30:
All the stock presets are very low volume level.

In order to use the full dynamic range of the output the output signal should peak at -18dB to -6dB (not clipping). This is very important for gain staging.

rerdavies, 2025-07-30:
They should. I'll take a look. Just checking, do you have your input levels
trimmed properly. I *thought* they were relatively 0db-ish input to output.
But it has been a very long time since I looked at them. All bets are off
if the input vu meters on the start nodes in your presets are not ALSO in
the -18db+ range. Asking only because you can probably check yourself
pretty quickly, and I'm in code atm so I can't check right away. Don't
bother to answer if you aren't there at the moment. I'll find out soon
enough.

rerdavies, 2025-09-07:
Default presets have been substantially reworked to make better use of plugins that have been added since the early days of PiPedal. They now have more-or-less accurate gain staging now.  (And there a substantial set of new default presets featuring NAM models that's in the works). 

To see the current default presets, take a look in the new "Factory Presets" bank that will be installed by PiPedal v1.5.90. (I can't replace existing presets).

closed 2025-09-07 by rerdavies

jardous, 2026-01-03:
After tuning a buffer to connect it directly to ADC I found, that it is the NAM models that need adjustments. I now use -12dB on input and -12dB on output.
The factory presets have the right volume when ADC gain and other audio interface values are set correctly.