rerdavies/pipedal #372: [BUG] CPU consumption inside WebUI correct?

SlayermanGR, 2025-08-16:
Greetings and congratulations for your work on PiPedal, such an amazing project!
I noticed there are some discrepancies between the Task Manager and PiPedal's WebUI regarding CPU consumption.

I have a preset that consumes ~30% in WebUI. In the Task Manager I can see PiPedal's task occupying ~15%.

Now there are some even weirder cases (which I haven't managed to steadily reproduce, but I'm working on it) where CPU consumption in WebUI for the same preset goes up (and stays up) to 60%-95% without touching anything in PiPedal, whereas the Task Manager still shows 15%. (No XRuns though!!)
Ways to get around this are a) unplug/replug the USB Scarlett sound card, b) open PiPedal settings, card/buffer settings and just click ok, c) restart PiPedal service.

Could you please share some insight on how CPU consumption is calculated inside the UX and if there is something that could cause CPU spikes?

** Environment (please complete the following information):**
 - OS: Raspberry Pi OS Latest Bookworm
 - Pi5 8GB

rerdavies, 2025-08-17:
## What's being measured.

The CPU use metric that PiPedal calculates measures how much CPU is being used by plugins (specifically) out of the total amount of time available to plugins (specifically) before an overrun will occur. It excludes system overhead from available time, and measures CPU use on one thread only -- on the the real-time audio thread.  Inasmuch as it is practically possible, you should be able to add plugins until PiPedal's CPU use indicator goes to 99%. Add one more percent, and you get XRUNs.  But (because of various sources of jitter in a heavily loaded system)), thinks get a little rough once you get above 90% CPU use. 

So not the same thing at all that htop, or btop, or whichever utility you are using to measure overall CPU use in the system. Most substantially different because they are measure 4 CPU cores, not one. And total CPU use sums to 400%, not 100%.  But also different because PiPedal's metric excludes system overhead, and concentrates only on time available to the plugins specifically.

The Pipedal CPU USE indicator is calculated as: 

     CPU USE =  100 * (PLUGIN CPU USE)/(PLUGIN CPU USE + TIME_SPENT_WAITING). 

And surprisingly (to ME, anyway), it's actually a fairly good indication of how many more plugin instances you can add before overruns occur. It depends on the plugins (which have to spend a constant amount of time processing each audio frame -- not always true), and it depends to certain extent on jitter in the system caused by background calculations, or servicing interrupts. But it is, actually possible to get up to.. at least 90% of CPU use without overruns occurring. Perhaps even higher. 

I would read your 30% / 15% example as follows:

- 30% of time available to plugins. Super! Needs more plugins! 
- After adding back all the overhead, and including system overhead for transferring audio to the USB device, that might actually work out to 60% of one single core being used.
- And 60% on one core (mostly the reatitme audio thread) works out to be about 15% of all 4 cores (which is what your system utility is reporting.

## Heavy CPU use on light loads.

The cases where you have  seen 70% or 80% CPU use on a lightly loaded system...  If I am not mistaken, what you are describing was (is?)  a bug that occured when you are using threading on TooB NAM. Present in the previous release, for sure. I'm hoping it's gone in the current release.  

A weird timing bug that happened because TooB NAM read the previous frame on the realtime thread, before it wrote the next frame. Doing those operations in the opposite order ensures that the real-time thread never stalls while waiting for data (as long as there is sufficient 4-core CPU power). Doing it in the wrong order locks into a cycle where the real-time thread ALWAYS waits for data after the first underrun; whereas doing it in the right order ensures that the reltime thread NEVER waits for dta.   Strange, subtle bug, hopefully gone. Let me know if you are still seeing it.

SlayermanGR, 2025-08-17:
Thank you so much for taking the time to write such a detailed response!

> But it is, actually possible to get up to.. at least 90% of CPU use without overruns occurring. Perhaps even higher.

I've seen 97%-98% CPU in PiPedal, yet no XRuns and real-time performance was good.

> and it depends to certain extent on jitter in the system caused by background calculations, or servicing interrupts

One thing I've noticed while trying to reproduce the heavy CPU issue is that it occurs when I'm doing other heavy-ish tasks while PiPedal is running the background (actually PiPedal is always running), but the reproduce pattern is not steady at the moment.

>  If I am not mistaken, what you are describing was (is?) a bug that occured when you are using threading on TooB NAM.

I am working with PiPedal's latest version (v1.4.87), so it's either not completely solved yet or maybe I've hit another wall. Also I am not using the "Threaded" functionality in TooB NAM, as it seems to introduce additional latency. Although, I do indeed remember when facing the heavy CPU issue that toggling Threaded on/off also actually seemed to solve the issue in the same way as the 3 other methods mentioned in my first post. Weird!

Thank you again for the prompt response and please let me know if I can provide you with any evidence/info that may prove useful for further investigation.

rerdavies, 2025-08-17:
Sorry. A very TLDR response. I have done my best to provide various points at which you can decide to read no further.

## Re: running at 97-98%... 

That is good news indeed! I have to admit, I have not pushed CPUs to the limit myself.  That would be accuracy above and beyond what I would have expected!

## first and foremost:

I'd like to collect this data before we go anywhere. 

- Which OS are you using? 
- What is your current buffer configuration? 
- Which plugins are you using that cause high CPU use? If Neural Amp Modeler, which NAM are you using? If convoution reverb, WHICH convolution reverb?
- Can you give me some indication of what sort of heavy processing you are doing on non-realtime threads? Ideally, you shouldn't be doing any heavy processing ... see notes later on. 

##  Plugins with stability problems at high load

You should categorically and definitely avoid using the Nam plugin distributed from Patch Storage, and similarly avoid use of anything derived from MOD NAM sources. Both have very dubious provenances. The Patch Storage NAM is derived from MOD NAM (or perhaps is perfectly identical with MOD NAM), and MOD NAM is a heavily patched version of  an ancient build of Mike Oliphant's NAM plugin, which has known problems with real-time loading. That family of source performs memory allocations on the realtime thread, which causes significant instability when used on a Realtime audio thread, along with severe intermittent overruns depending on the state of the heap. And, furthermore, the MOD NAM patches (which implement -- among other things --  the Buffer switch) will cause memory corruption when the Buffer switch is  turned on, and XRUNs occur. Avoid at all costs. 

Either use TooB NAM instead, or use a freshly compiled version of Mike's NAM from Mike Oliphant's Github repository. Both use the same backend NAM implementation from Mike Oliphant's MIT-licenced NueralAudio project, although Toob NAM adds the threading feature. Kudos to Steve Atkinson and Mike Oliphant who have done all the hard work in TooB NAM's current incarnation.

I have sent a note to the author of the PatchStorage NAM asking him to update to current sources, but he will lose the Buffer feature (which is distributed through a MOD patch that has a CC BY NC ND license of all things!  No Derivatives! A license that is actually incompatible with Mike's GPL-3 license). And I have yet to receive any response. As I said, a very sketchy provenance! 

I *THINK* that Ratatouille is probably ok, if you're looking for a NAM plugin other than TooB NAM (for some reason),  that supports threading/buffering. At least, it does not trace heritage through the MOD NAM patches. And it is likely to have picked up malloc fixes from whichever NAM backend it is currently using.  (Assuming that the PatchStorage version of Ratatouille is actively maintained, and has been updated in the last 9 months or so). 

I would similarly be cautious about using Convolution Reverbs that trace their heritage to MOD plugins, as those may replicate the memory corruption bug found in MOD NAM.  (A similar problem - marshalling audio data to and from the realtime thread, which may use the same code as MOD NAM). 

## How to recover

If the issue isn't related to a complete audio hang, or ALSA getting completely confused, what you would actually need to do is force a reload of the currently running plugins. 

A semi-documented feature: plugins do not get reloaded if you switch between presets that have the  identical plugin structure. If the structure is identical, you get the same sort of handling that a Snapshot change would get: control values change, but the plugins themselves do not get reloaded.  You have to force the plugins to get reloaded; and the best way to force them to get reloaded is to load a preset with a very different structure (which plugins, in which order, how connected). Or drag one of the plugins after or before another plugin on the pedalboard. I am not sure, off the top of my head, whether blank slots cound in the calculation of "Identical structure". I think they do not. 

If this actually works, I would strongly suspect an issue with one of the plugins, rather than with PiPedal, or ALSA.

## Get the latest OS updates. 

You should definitely make sure your system is fully updated. Upstream work in pipewire+ALSA seems to be pretty active -- with ongoing fixes to both ALSA and pipewire relating to real-time performance bugs. Raspberry Pi OS actually does a fairly good job of keeping up with upstream fixes for the audio stack; Ubuntu, seems to backport less aggressively (although there have been recent updates to ALSA/pipewire in both 24.04, and 24.10.  And massive changes in 25.04 (currently a research project, as it will require a separate release binary from 24.04/24.10/Raspberry Pi OS because of a stupidity relating to LIBZIP4, of all things). 

## Doing heavy-ish tasks in non-realtime.

The short answer to that may be: if you ARE loading the Pipedal server with non-realtime work, you WILL have to increase buffer sizes to ensure that XRuns don't occur.  Perhaps to unacceptably large sizes from a latency perspective. Experiment, and see what the smallest buffer size you can get away with is. 

There are outstanding problems when recovering from heavy overruns, caused by excessive non-realtime activity on the server. They are proving to be extremely difficult to debug. Not least because I think there is a watchdog process that ki [... cut ...] ad. I am HUGELY interested in any patterns you may spot if that happens (which plugins, especially). If the CPU use sticks at a very high value (typically 80%+), and does not change at all after that, then you have run into this problem. This is a big deal. And any clues you can spot would be enormously useful.  

Again: which OS, which plugins, what buffer configuration, any patterns in what's going on in your non-realtime code that you happen to notice  would be the data that would be helpful.

rerdavies, 2025-08-21:
>  Although, I do indeed remember when facing the heavy CPU issue that toggling Threaded on/off also actually seemed to solve the issue in the same way as the 3 other methods mentioned in my first post. Weird!

Just noticed that. That's an interesting clue that may lead to a fix. Toggling threading doesn't do anything other than restart the foreground/background queues, and provide ONE frame of relief from heaving computing. I will review code with that observation in mind.

rerdavies, 2025-08-21:
Fascinating!  :-)

So. Point number one... Don't do that! lol. To be honest, I am rather amazed that Pipedal works at all in that situation. It would not remotely work on a Pi 4.  

I think what you're seeing is PiPedal doing its best to run when teetering on the verge of running out of total available computing power on all 4 CPUs.  because it has to compete with realtime-priority computing being done by PiPewire in order to generate video and audio from the YouTube video. Rendered browser pages that don't have audio or video may also travel through the pipewire rending pipeline as well, although I can't say for sure that's true.  

I don't know exactly what the pipewire realtime thread does or does not do. But it is running with effectively maximum possible realtime priority for exactly the same reason that pipedal does -- so that it can deliver xrun-free audio and video.

While the behaviour is a bit odd (and a bit hard to explain), Pipedal is actually doing exactly what it should do under the circumstances, and what it needs to do to deliver the last possible sliver of compute performance. 

If you see that sort of behaviour while not running a browser instance, that would be a bug (-ish), depending on how heavily you have loaded the CPU. If running close to maximum then the behaviour is actually a "feature" not a "bug". 

Playing Youtube videos is possibly the worst possible thing you could do while running PiPedal.  Certainly, all of the audio playback  for the video ends up running at realtime priority. I would imagine that the final stage of rendering video to screen also occurs at realtime priority as well  I'm not sure what the exact thread priorities are, but they look approximately like this. (The choice of priorities relative to `pipewire` is very deliberate).

- RT Priority 99: The Realtime watchdog process, running at maximum possible priority.
- RT Priority 96: Pipedal realtime thread, including ALSA operations.
- RT Priority 95: Pipewire audio video  and MIDI realtime thread. 
- RT Priority 93: Background processing for the NAM instances.
-  RT priority 60-ish: Hardware interrupt service threads!
- RT priority 50-ish: Hardware-related realtime operations.

Add, as an added complication, the realtime NAM instances will run slower when lower-priority threads are making heavy use of L2 and L3 cache (because most of the NAM computing operations are actually more memory-intensive than they are compute intensive).  And decoding video (or probably, rendering a busy browser page) is exactly the sort of operation that's going to poison the Pipedal realtime audio threads' data caches. Not a big hit, but certainly a few extra percentage points of CPU use.

So. When running threaded NAM instances, when everything is healthy, the realtime audio thread never has to wait for data. Audio data for the foreground code is just sitting there ready to be transferred the audio device. If things are running behind, then the foreground thread has to make a kernel call to wait until that data does become ready, and a another kernel call has to be made to get the realtime  thread running again once data does become available . That kernel modestly increases the total compute cost (both on the realtime threaded, and an increase in computing cost on all four CPUs, so it is a bit stressful for the overall system. And although the time spent waiting doesn't contribute to overall system computing time, any time spent waiting counts directly and entirely toward the CPU Use figure that PiPedal reports, since time spent waiting reduces the time that the realtime thread has to run the entire effect chain. 

The scheduling transitions are modestly expensive -- an additional 250 microseconds of CPU use on a Pi 4 perhaps a few percent of total. Maybe a couple of % of CPU time on the realtime thread, and maybe a couple of % of overall computing available on all four CPUS. 

So. The bump in PiPedal's CPU use occurs because the second NAM instance is running slightly behind the curve, and has to actually wait for data to become available. That data does arrive, so there's no XRun; HOWEVER the realtime audio thread is spending 30-70% of its available time waiting for that data to arrive. So it displays high CPU use. 

And once it has waited once, all subsequent calculations take slightly longer because of the additional scheduler transitions in the kernel.  So the behaviour is sticky. Waiting once causes all subsequent frames to wait too (because overrall 4-CPU compute time has increased very slightly because of the addition scheduler transitions).

Due to a quirk in the handoff of the NAM instance between foreground and background thread,  one frame of audio actually gets dropped. So that SAVEs a tiny sliver of overall computing costs, and that is probably enough to ensure that subsequent realtime thread operations break out of "always wait for data" mode, and flip back to "never wait for data" mode. And PiPedal's CPU use drops back to normal again.

So. Just to summarize. PiPedal is not designed to be used when a browser is running on the Pi.  ideally, you should be using a remote browser running either on your phone, or a remote desktop.  And you should NEVER be using pipedal while playing YouTube videos.  

The bi-stable behavior where PiPedal flips from "never wait for data" mode (low PiPedal CPU Use reported) to "always wait for data" mode (high PIpedal CPU use reported, but actually about the same 4-CPU compute cos [... cut ...]
While the bi-stable behaviour (gettings "stuck" in wait-for-data mode) is a bit odd, I'm not honestly sure that I could fix the behaviour even if I wanted to. When not competing with a browser for available realtime scheduling time, the behaviour is a "feature", not a "bug", and is what PiPedal has to do in order to make use of the last few percentage points of overall 4-CPU compute time.

And it's definitely interesting data that may come in handy in future, if there do turn out to be bugs in the way this feature works. 

Thanks for reporting it.