rerdavies/pipedal #375: how to debug doesn't work in ubuntu

hmollercl, 2025-08-21:
Installed in ubuntu 24.04 ( `sudo apt-get install ./pipedal_1.4.87_amd64.deb` ) w/o problems but it doesn't seem to be running even after reboot. http://localhost:81 doesn't respond (although localhost:80 respond)  and I don't see it running with top. Tried `pipedalconfig --start` and `--restart` but nothing seems to work. What should I look for to find the problem?

rerdavies, 2025-08-21:
Do a reboot, and run the following command:

    journalctl -b 0 | grep pipedald

Post the results here, and I'll take a look to see what's going on.

hmollercl, 2025-08-22:
```
Aug 21 20:35:27 lenovoKubuntu systemd[1]: Starting pipedald.service - PiPedal Web Service...
Aug 21 20:35:27 lenovoKubuntu pipedald[1509]: Document root: /etc/pipedal/config Threads: 5
Aug 21 20:35:27 lenovoKubuntu pipedald[1509]: Locale: es_CL
Aug 21 20:35:28 lenovoKubuntu pipedald[1509]: Checking for updates.
Aug 21 20:35:28 lenovoKubuntu pipedald[1509]: Scanning for LV2 Plugins
Aug 21 20:35:28 lenovoKubuntu pipedald[1509]: Updater: Failed to fetch update info. Server has no internet access.
Aug 21 20:35:28 lenovoKubuntu pipedald[1509]: Updater: Next update check at 2025-08-22 00:35:28
Aug 21 20:35:38 lenovoKubuntu systemd[1]: Started pipedald.service - PiPedal Web Service.
Aug 21 20:35:38 lenovoKubuntu pipedald[1509]: No ALSA device selected.
Aug 21 20:35:39 lenovoKubuntu pipedald[1509]: Failed to start audio. Audio configuration not valid.
Aug 21 20:35:39 lenovoKubuntu pipedald[1509]: Audio started. DUMMY, n/a, Native float, 48000, 64x3, in: 2/2, out: 2/2
Aug 21 20:35:39 lenovoKubuntu pipedald[1509]: Listening on mDns address lenovoKubuntu:81
Aug 21 20:35:39 lenovoKubuntu pipedald[1509]: WebServer: asio listen error: system:1 (Operation not permitted)
Aug 21 20:35:39 lenovoKubuntu pipedald[1509]: Web server: Underlying Transport Error
Aug 21 20:35:39 lenovoKubuntu pipedald[1509]: DNS/SD service announced. (pipedal)
Aug 21 20:35:40 lenovoKubuntu pipedald[1509]: DNS/SD group established.
Aug 21 20:35:40 lenovoKubuntu pipedald[1509]: HotspotManager: Monitoring network status.
Aug 21 20:35:50 lenovoKubuntu pipedald[1509]: Listening on mDns address lenovoKubuntu:81
```

hmollercl, 2025-08-24:
I installed in another kubuntu machine and it was installed listening in port 80 (instead of 81) in this installation that wasn't a problem, but maybe the way that you discover if it is raspberry or ubuntu doesn't work well in kubuntu??. Maybe the whole problem was this, it tried to start in port 80 and couldn't.

rerdavies, 2025-08-24:
The test is pretty simple: make a web request to port 80, and if there's a response, try port 81, (and then port  ....8080? I'm not sure exactly what happens after port 81).  If you have any other ideas as to what went wrong, do let me  know.

rerdavies, 2025-09-07:
Feel free to re-open if you see the issue again.

closed 2025-09-07 by rerdavies