rerdavies/pipedal #380: [BUG] No suitable devices with pisound on Raspberry Pi 4

kidproquo, 2025-08-30:
**Describe the bug**
The web UI shows "No suitable devices" when trying to configure the audio device

**Screenshots**

[image]

`pipedalconfig` shows the device: 
```
pipedalconfig --alsa-devices
=== ALSA Hardware Devices ===

Sound Card hw:pisound
  Name: pisound
  Driver: pisound

  Subdevice: hw:3,0 - hw:3,0
  ID: PS-1ZJQH40 snd-soc-dummy-dai-0
  Capabilities: PLAYBACK CAPTURE
  Channels: 2 - 2
  Sample rates: 48000 - 192000 Hz
  Supported formats: S16_LE S24_LE S32_LE
  Channel maps:
       No channel maps.
```
```
pipedalconfig --alsa-device hw:pisound
Device: hw:pisound
Channel range: 2 to 2

Supported channel configurations:
  2 channels (Stereo)
```
```
Logs:
Aug 29 14:49:50 pi-riffpod-001 systemd[1]: Starting pipedald.service - PiPedal Web Service...
Aug 29 14:49:51 pi-riffpod-001 pipedald[691]: Document root: /etc/pipedal/config Threads: 5
Aug 29 14:49:51 pi-riffpod-001 pipedald[691]: Locale: en_US
Aug 29 14:49:51 pi-riffpod-001 pipedald[691]: Updater: Next update check at 2025-08-29 22:31:30
Aug 29 14:49:51 pi-riffpod-001 pipedald[691]: Scanning for LV2 Plugins
Aug 29 14:49:52 pi-riffpod-001 systemd[1]: Started pipedald.service - PiPedal Web Service.
Aug 29 14:49:52 pi-riffpod-001 pipedald[691]: No ALSA device selected.
Aug 29 14:49:52 pi-riffpod-001 pipedald[691]: CrashGuard: Detected previous crash, count = 1
Aug 29 14:49:54 pi-riffpod-001 pipedald[691]: Failed to start audio. Audio configuration not valid.
Aug 29 14:49:54 pi-riffpod-001 pipedald[691]: Audio started. DUMMY, n/a, Native float, 48000, 64x3, in: 2/2, out: 2/2
Aug 29 14:49:54 pi-riffpod-001 pipedald[691]: Listening on mDns address pi-riffpod-001:80
Aug 29 14:49:54 pi-riffpod-001 pipedald[691]: DNS/SD service announced. (pipedal)
Aug 29 14:49:55 pi-riffpod-001 pipedald[691]: DNS/SD group established.
Aug 29 14:49:55 pi-riffpod-001 pipedald[691]: HotspotManager: Monitoring network status.
Aug 29 14:50:05 pi-riffpod-001 pipedald[691]: Listening on mDns address pi-riffpod-001:80
Aug 29 14:50:05 pi-riffpod-001 pipedald[691]: Listening on Wi-Fi address 192.168.1.4:80
```

** Environment (please complete the following information):**
- OS: Debian Bookworm 64-bit
- if Linux, run `uname -a` and include the results. - `Linux pi-riffpod-001 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux`
 - Device that the pipedal server is running on (pi 3, pi 4, &c, how much system memory) - Raspberry Pi 4, 4GB RAM
 - PiPedal Version - 1.4.87 Beta
If UI-related:
 - Firefox browser on Mac to access the web UI (`http://IP address of Pi`)

rerdavies, 2025-08-30:
Is it possible that something else is using the device? PiPedal needs
exclusive access.

Also worth checking after a reboot, to see if that helps.

jardous, 2026-01-03:
@kidproquo did you solve the issue? make sure no other processes are using the device.