rerdavies/pipedal #387: After installing the update, then uninstalling and reinstalling PiPedal, I'm still stuck in a service crash/restart loop.

rerdavies, 2025-09-08:
After installing the update, then uninstalling and reinstalling PiPedal, I'm still stuck in a service crash/restart loop.
What information can I provide that will help diagnose this issue?

_Originally posted by @DuraMorte in https://github.com/rerdavies/pipedal/discussions/385#discussioncomment-14334198_

rerdavies, 2025-09-08:
What i need is logjournal resuls from when the service crashes. 

Install the new 1.5.90 build, reboot, and (after the service crashes a couple of times), run:

```
     journalctl -b 0 | grep pipedald
```

DuraMorte, 2025-09-08:
Sep 07 22:21:06 ryan-pipedal systemd[1]: pipedald.service: Scheduled restart job, restart counter is at 1.
Sep 07 22:21:06 ryan-pipedal systemd[1]: Starting pipedald.service - PiPedal Web Service...
Sep 07 22:21:06 ryan-pipedal pipedald[3013]: Document root: /etc/pipedal/config Threads: 5
Sep 07 22:21:06 ryan-pipedal pipedald[3013]: Locale: en_US
Sep 07 22:21:06 ryan-pipedal pipedald[3013]: Updater: Next update check at 2025-09-08 02:52:15
Sep 07 22:21:06 ryan-pipedal pipedald[3013]: Scanning for LV2 Plugins
Sep 07 22:21:06 ryan-pipedal systemd[1]: Started pipedald.service - PiPedal Web Service.
Sep 07 22:21:06 ryan-pipedal pipedald[3013]: Found ALSA device hw:U192k.
Sep 07 22:21:06 ryan-pipedal systemd[1]: pipedald.service: Main process exited, code=dumped, status=11/SEGV
Sep 07 22:21:06 ryan-pipedal systemd[1]: pipedald.service: Failed with result 'core-dump'.
Sep 07 22:21:11 ryan-pipedal systemd[1]: pipedald.service: Scheduled restart job, restart counter is at 2.

DuraMorte, 2025-09-08:
I feel like I might need to back up a bit to provide any useful information...

Sep 07 22:20:51 ryan-pipedal pipedald[1842]: WebServer: Error getting remote endpoint: system:9 (Bad file descriptor)
Sep 07 22:20:51 ryan-pipedal pipedald[1842]: WebServer: asio async_shutdown error: system:9 (Bad file descriptor)
Sep 07 22:20:51 ryan-pipedal pipedald[1842]: WebServer: handle_accept error: Operation canceled
Sep 07 22:20:51 ryan-pipedal pipedald[1842]: WebServer: Stopping acceptance of new connections because the underlying transport is no longer listening.
Sep 07 22:20:52 ryan-pipedal pipedald[1842]: DNS/SD service stopped.
Sep 07 22:20:52 ryan-pipedal pipedald[1842]: Audio stopped.
Sep 07 22:20:52 ryan-pipedal pipedald[1842]: Audio thread terminated.
Sep 07 22:20:52 ryan-pipedal pipedald[1842]: Audio processing terminated unexpectedly.
Sep 07 22:20:52 ryan-pipedal pipedald[1842]: Stopping web server.
Sep 07 22:20:55 ryan-pipedal pipedald[1842]: WebServer: failed to close all connections.
Sep 07 22:20:55 ryan-pipedal pipedald[1842]: WebSocketSession closed. [::ffff:192.168.1.74]:62020
Sep 07 22:20:55 ryan-pipedal pipedald[1842]: Web server terminated.
Sep 07 22:20:55 ryan-pipedal kernel: ppdl_dbusd[1860]: segfault at 7ff05b7ae524 ip 0000559554326edf sp 00007ff7b6ffca20 error 4 in pipedald[1e0edf,559554179000+24d000] likely on CPU 2 (core 0, socket 0)
Sep 07 22:20:55 ryan-pipedal systemd[1]: pipedald.service: Main process exited, code=dumped, status=11/SEGV
Sep 07 22:20:55 ryan-pipedal systemd[1]: pipedald.service: Failed with result 'core-dump'.
Sep 07 22:20:55 ryan-pipedal systemd[1]: Stopped pipedald.service - PiPedal Web Service.
Sep 07 22:20:55 ryan-pipedal systemd[1]: pipedald.service: Consumed 32.240s CPU time.
Sep 07 22:21:00 ryan-pipedal systemd[1]: Starting pipedald.service - PiPedal Web Service...
Sep 07 22:21:00 ryan-pipedal pipedald[2750]: Document root: /etc/pipedal/config Threads: 5
Sep 07 22:21:00 ryan-pipedal pipedald[2750]: Locale: en_US
Sep 07 22:21:00 ryan-pipedal pipedald[2750]: Updater: Next update check at 2025-09-08 02:52:15
Sep 07 22:21:00 ryan-pipedal pipedald[2750]: Scanning for LV2 Plugins
Sep 07 22:21:00 ryan-pipedal systemd[1]: Started pipedald.service - PiPedal Web Service.
Sep 07 22:21:00 ryan-pipedal pipedald[2750]: Found ALSA device hw:U192k.
Sep 07 22:21:01 ryan-pipedal systemd[1]: pipedald.service: Main process exited, code=dumped, status=11/SEGV
Sep 07 22:21:01 ryan-pipedal systemd[1]: pipedald.service: Failed with result 'core-dump'.
Sep 07 22:21:06 ryan-pipedal systemd[1]: pipedald.service: Scheduled restart job, restart counter is at 1.

rerdavies, 2025-09-08:
No. That's good enough. I think that maarthome may have giving me what I
need. Hold that thought, and I'll get back to you.

rerdavies, 2025-09-08:
I have reproduced the issue. Fix forthcoming.

DuraMorte, 2025-09-10:
Fix successful. System is up and working flawlessly.